n0bel/PiClock #110: Move to another mapping platform?

n0bel, 2018-12-21:
I'm taking input on using a different mapping provider to get rid of the google maps credit card requirement.

n0bel, 2018-12-21:
MapBox seems to have an equivalent API to google static maps.  After making a free account you can play with it here:  https://www.mapbox.com/help/static-api-playground/    The url and parameters are different but close enough.   Looks like a drop in replacement.
Most of the work has already been done here: https://github.com/n0bel/PiClock/pull/84

n0bel, 2018-12-23:
Debugging the MapBox version ... Available on the dev-mapbox branch
Changes:
Directly paint Markers from local images, to allow for easy custom markers
Implemented MapBox optionally (enabled in Config and ApiKeys)

[image]
[image]

merky1, 2018-12-25:
Have you seen much of a difference, quality wise in the map images?

BertLindeman, 2018-12-25:
Just an example on my PC screen. Looks fine to me after annoying look at a gray background provided by google ;-)

[image]

n0bel, 2018-12-26:
They are different.   Google was more terrain-map-ish and satellite combo, where mapbox is more straight satellite.   I've not tried all the mapbox maps.

n0bel, 2018-12-26:
mapbox is now live on the master branch.  It is optional.

BertLindeman, 2018-12-26:
An example html page to see how **your** map would look like?
Use your preferred location / zoom / size,
and copy the access token from the example(s) on the mapbox documentation page:
```https://www.mapbox.com/api-documentation/#styles```
A well willing freely usable webpage by me. Use it as you like and no guarantee:

```
<!DOCTYPE html>
<html lang="en">
<head>
<title> mapbox style TEST</title>
</head>
<body>
<script>
var urlhome     = "https://api.mapbox.com/styles/v1/mapbox/";

/* Take the key from one of the examples on URL 
   https://www.mapbox.com/api-documentation/#introduction */
var key = "?access_token=pk.eyJ1IjoibWFwYm94IiwiYSI6ImNpejY4NDg1bDA1cjYzM280NHJ5NzlvNDMifQ.d6e-nNyBDtmQCVwVNivz7A";

function showExample(loc, zoom, size, style) {
    var urlzoom     = zoom; 
	    urlzoom    += ",0,0/";
	var urlsize     = size;	
	    urlsize    += "/";	
	var urlstyle    = style;         
	    urlstyle   += "/static/"; 	
	var url  = urlhome;
		url += urlstyle;
		url += urllocation;
		url += urlzoom;
		url += urlsize;
		url += key;
	document.write("<h4>Style:"+style+" zoom:"+zoom+" size:"+size+"</h4><img id='"+style+"' src=");
	document.write(url+">");
};
/* Currently documented:
	streets-v10
	outdoors-v10
	light-v9
	dark-v9
	satellite-v9
	satellite-streets-v10
	navigation-preview-day-v4
	navigation-preview-night-v4
	navigation-guidance-day-v4
	navigation-guidance-night-v4
*/
var urllocation = "4.627072,51.892625,";
var zoom     = "4"; 
var size     = "166x146";
var style    = "streets-v10";      showExample(urllocation, zoom, size, style);
    style    = "outdoors-v10";     showExample(urllocation, zoom, size, style);
	style    = "light-v9";         showExample(urllocation, zoom, size, style);
    style    = "dark-v9";          showExample(urllocation, zoom, size, style);
	style    = "satellite-v9";     showExample(urllocation, zoom, size, style);
	style    = "satellite-streets-v10";        showExample(urllocation, zoom, size, style);
	style    = "navigation-preview-day-v4";    showExample(urllocation, zoom, size, style);
	style    = "navigation-preview-night-v4";  showExample(urllocation, zoom, size, style);
	style    = "navigation-guidance-day-v4";   showExample(urllocation, zoom, size, style);
	style    = "navigation-guidance-night-v4"; showExample(urllocation, zoom, size, style);
	
var zoom     = "7"; 
var size     = "255x255";
var style    = "streets-v10";      showExample(urllocation, zoom, size, style);
    style    = "outdoors-v10";     showExample(urllocation, zoom, size, style);
	style    = "light-v9";         showExample(urllocation, zoom, size, style);
    style    = "dark-v9";          showExample(urllocation, zoom, size, style);
	style    = "satellite-v9";     showExample(urllocation, zoom, size, style);
	style    = "satellite-streets-v10";        showExample(urllocation, zoom, size, style);
	style    = "navigation-preview-day-v4";    showExample(urllocation, zoom, size, style);
	style    = "navigation-preview-night-v4";  showExample(urllocation, zoom, size, style);
	style    = "navigation-guidance-day-v4";   showExample(urllocation, zoom, size, style);
	style    = "navigation-guidance-night-v4"; showExample(urllocation, zoom, size, style);

</script>
</body>
</html>

```

BertLindeman, 2018-12-26:
Additional thought: maybe also take the style from Config.py?
That would also allow for day / night style to be used if one uses the switch.py.

n0bel, 2018-12-26:
yup i'll add the style in config

n0bel, 2018-12-26:
https://www.mapbox.com/studio/
Play more!

n0bel, 2018-12-26:
Radar configs now allow for specifying mapbox styles.  As of commit: https://github.com/n0bel/PiClock/commit/cc24bad7a98131e45c032e33f7424068edf8774d   Default style is still 'mapbox/satellite-streets-v10'

BertLindeman, 2018-12-26:
Thank you! 
Did a pull and will fiddle with the radar settings here.

cranesoft, 2018-12-27:
Late to the party but I've been following the thread re: migration from WU.  @n0bel and others, thanks for all of the work to make it happen!

Everything works after cloning and migrating my old config files except the radar images.  Also, I have API keys setup for both MapBox and DarkSky.  When initially painted, I see what appears to be the satellite image (looks good) and then the current radar image overlays covering up the satellite image...just gray with radar image data.  Perhaps a transparency issue?  In hopes that pictures will better convey what I'm talking about I've attached an overall view of my clock as well as a detailed view of the two radar images mentioned above.  Anyone else see the same issue?  TIA!

[image]
[image]

cranesoft, 2018-12-27:
Good suggestions!  No errors in the log.  The calls to mapbox look good (copied to the browser as you suggested).  The calls to rainviewer.com may be good but I'm wondering if the white part (shows as black in the image below) of the image should be treated as transparent?

[image]

BertLindeman, 2018-12-27:
Tx. I took one radar image URI from my log and it has like yours a black background.
"My URL":
```
 https://tilecache.rainviewer.com/v2/radar/1545917400//256/7/65/41.png?color=3
```

Tried to paste the image here, but it refuses this time....
Reports: `Something went wrong`

BertLindeman, 2018-12-27:
Huh? The radar image above I "did" in Chrome. Another one I loooked at in Firefox and that has a white background. Sorry to add to the confusion.
```
 https://tilecache.rainviewer.com/v2/radar/1545920400//256/11/1051/678.png?color=3
```
No paste again..

BertLindeman, 2018-12-27:
Verified that the image **does* have an alpha channel (So has transparency). 

You **do** see the mapbox calls in your log. 
And that should mean those images **are** used.

Just to verify: do you have usemapbox in your active config:
```
usemapbox = 1   # Use Mapbox.com for maps, needs api key (mbapi in ApiKeys.py)
```
And re-reading your comment: do you still have radar definitions in your Config.py?

cranesoft, 2018-12-27:
I have Mapbox enabled in my config (usemapbox = 1) and I have radar1 through radar4 defined as well.

Upon a closer view of the new Config-Example.py I see that by default radar1 through radar4 have 'satellite' = 0 where I was setting those values to 1.  I changed my config to reflect the example and all is good with the universe again!  Thanks!

BertLindeman, 2018-12-27:
A configuration setting that is easy to misunderstand due to the recent changes ;-) 
Glad you got it resolved.

n0bel, 2018-12-28:
There's still one thing wrong....  The clock is crooked on your wall!

cranesoft, 2018-12-28:
If you tilt your head it IS level! ;^)

feh123, 2018-12-29:
Just setup with api keys from Dark Sky and Mapbox for Cambridge UK.  Works brilliantly with default settings. Thanks!

closed 2019-01-12 by n0bel