n0bel/PiClock #111: moonphase calculation has a small error

BertLindeman, 2018-12-23:
The calculation of the moon phase in the current PyQtPiClock.py:

-  has a typo in the value for ```Waxing Crescent``` the value ```0.625``` should be ```0.0625```
-  duplicate `if` for ```Waning Gibbous```

**Current code**
```
def phase(f):
    pp = 'New Moon'
    if (f > 0.625):
        pp = 'Waxing Crescent'
    if (f > 0.1875):
        pp = 'First Quarter'
    if (f > 0.3125):
        pp = 'Waxing Gibbous'
    if (f > 0.4375):
        pp = 'Full Moon'
    if (f > 0.5625):
        pp = 'Waning Gibbous'
    if (f > 0.5625):
        pp = 'Waning Gibbous'
    if (f > 0.6875):
        pp = 'Third Quarter'
    if (f > 0.8125):
        pp = 'Waning Crecent'
    if (f > 0.9375):
        pp = 'New Moon'
    return pp
```

As this part of the code tests all values all the time, I also propose to reverse the ```if```s like this:

```
**Proposed code**

def phase(f):
    # Determine moon phase upon the decimal part of the lunation number
    # Darksky points in their doc to URL:
    # https://en.wikipedia.org/wiki/Lunation_Number
    # the values here are mid-values halfway between these ranges:
    # 0 for new moon, 0.25 for first quarter moon, 0.5 for full moon, and 0.75 for last quarter moon.
    if   (f > 0.9375):
            pp = 'New Moon'
    elif (f > 0.8125):
            pp = 'Waning Crecent'
    elif (f > 0.6875):
            pp = 'Third Quarter'
    elif (f > 0.5625):  ## BUG: was duplicate IF
            pp = 'Waning Gibbous'
    elif (f > 0.4375):
            pp = 'Full Moon'
    elif (f > 0.3125):
            pp = 'Waxing Gibbous'
    elif (f > 0.1875):
            pp = 'First Quarter'
    elif (f > 0.0625):  ## BUG: Zero added for this if was 0.625
            pp = 'Waxing Crescent'
    else:
            pp = 'New Moon'
    return pp
```

I will add a separate enhancement request to take the moon phase text from the Config.py

n0bel, 2018-12-23:
Great catch great work...
Would you like to do a pull request?

BertLindeman, 2018-12-23:
Thanks! Just enough magic words to get me started.

BertLindeman, 2018-12-23:
Looks like i made it ;-)

n0bel, 2018-12-23:
Wonderful!

n0bel, 2018-12-23:
Might as well do the other one

BertLindeman, 2018-12-23:
Now I have the grip of it hihi

n0bel, 2018-12-23:
@BertLindeman  Thanks!   All merged, plus a few other additions..   Great Work!

closed 2018-12-23 by n0bel