n0bel/PiClock #120: API’s not working.

grayeagle57, 2019-01-06:
I did this project 2 years ago. Lately the map and radar section haven’t been working. I attempted the updated build but ran into issues where API keys are installed. I am not familiar with python or writing code. Having said that, I probably do not understand the instructions for putting the keys into the file. I put them in between the quotation marks as indicated. Nothing happens. So Any help would be appreciated in helping me to get this fantastic clock working again

n0bel, 2019-01-06:
I'll give  you some general things.

1. Check the log.    Maybe there are some hints there.
```
cd PiClock/Clock
cat PyQtPiClock.1.log
```
2. Review the update instructions.
https://github.com/n0bel/PiClock/blob/master/Documentation/Install.md#updating-to-newerupdated-versions

3. Review the instructions for setting up API Keys.
https://github.com/n0bel/PiClock/blob/master/Documentation/Install.md#configure-the-piclock-api-keys

4. Possibly start over with a new api key file.
```
cd PiClock/Clock
cp cp ApiKeys-example.py ApiKeys.py
nano ApiKeys.py
```
Put in your API keys as indicated. 
A sample completed ApiKeys.py file might look like the following.
Note that the keys given won't work, they are just examples.
Note that I omitted the Google Maps API since this example uses Mapbox.   So its just an empty string (two single quotes)
```
#change this to your API keys
# DarkSky API key
dsapi = '498135d8b3c3f7c8d02fc12c8ac01b55f257c176 '
# Map API keys -- only needs 1 of the following
# Google Maps API key (if usemapbox is not set in Config)
googleapi = ''
# Mapbox API key (access_token) [if usemapbox is set in Config]
mbapi = 'pk.VBTLUhMbnDuxdzXfJHlgCGpqkviQOjWtryYcIRNsmAZawFSPKoJHlgCGpqkviQOj.MbnDuxdzX'
```
5. The last set of changes were made on December 26th.   Most are minor.   One had to do with the 'satellite' flag, which is now removed but was causing some people some issues.   If you've not updated to the latest version, you'll want to do that.

Hope this helps.

grayeagle57, 2019-01-08:
Well I have the clock, radar looks like it’s trying to work , but no maps or weather data.

  1.  Review the update instructions.
https://github.com/n0bel/PiClock/blob/master/Documentation/Install.md#updating-to-newerupdated-versions

  2.  Review the instructions for setting up API Keys.
https://github.com/n0bel/PiClock/blob/master/Documentation/Install.md#configure-the-piclock-api-keys

  3.  Possibly start over with a new api key file.

  1.  The last set of changes were made on December 26th. Most are minor. One had to do with the 'satellite' flag, which is now removed but was causing some people some issues. If you've not updated to the latest version, you'll want to do that.

Hope this helps.

togatown, 2019-01-08:
Did you sign up for the two new APIs?

DarkSky has replaced Weather Underground and Mapbox has replaced Google Maps
[https://darksky.net/dev](https://darksky.net/dev)
[https://www.mapbox.com/signup/](https://www.mapbox.com/signup/)

If you haven't heavily customized the clock, I would rename the PiClock folder and start again with a fresh install instead of an update. 

Since you've already had it running once, skip right to the [Get the PiClock Software](https://github.com/n0bel/PiClock/blob/master/Documentation/Install.md#get-the-piclock-software) section.

Then skip to the [Configure the PiClock api keys](https://github.com/n0bel/PiClock/blob/master/Documentation/Install.md#configure-the-piclock-api-keys) section
DarkSky is required. Google Maps and Mapbox or an either/or, choose one but I would use Mapbox.

Then jump down to the [Configure your PiClock](https://github.com/n0bel/PiClock/blob/master/Documentation/Install.md#configure-your-piclock) section

You should be able to skip the rest. This should at least get you running and then you can change options in your config to match your old one in the renamed folder. You may find it more comfortable to edit the apikeys and config files in a text editor instead of the nano command line editor. Check out this [link](https://itsfoss.com/notepad-alternatives-for-linux/)

grayeagle57, 2019-01-10:
After I put the API keys into the file, what’s supposed to happen? When I do that do then  go straight to “configure “? Either way nothing happens. If I close the file editor and then go to “configure”, doesn’t that erase the API’s?

https://darksky.net/dev
https://www.mapbox.com/signup/

togatown, 2019-01-10:
> After I put the API keys into the file, what’s supposed to happen?

Did you start from scratch by shutting down the clock, renaming the PiClock folder and getting a fresh copy or are you still working from the updated copy?

>  If I close the file editor and then go to “configure”, doesn’t that erase the API’s?

Are you using the nano command line editor to put the API keys in? If so, are you saving the file before exiting?

togatown, 2019-01-12:
Check my profile and send me an email. It'll be easier to communicate

n0bel, 2019-01-12:
The best place to start is to see what is showing errors in the log file, as given in the first comment.

It can also log to the terminal as shown in the instructions here: https://github.com/n0bel/PiClock/blob/master/Documentation/Install.md#run-it

n0bel, 2019-01-12:
@togatown Thanks for the help!

grayeagle57, 2019-01-12:
I'm sorry I just don't know how to do that.

grayeagle57, 2019-01-12:
This is what I found: libpng warning: known incorrect sRGB profile. Also: ValueError: NO JSON object could be decoded.
Let me emphasize that I really appreciate your help. But I know nothing about Python or writing/ understanding code.

n0bel, 2019-01-13:
There should be more in the log file..   
NO JSON object could be decoded.
Its either a problem with your api key for darksky or a problem with your latitude and longitude for the current conditions.    Just above the error, you'll see a url to darksky.net.   It'll be best if we can see that.   Hide/scramble the portion that shows the api key.

grayeagle57, 2019-01-13:
https://api.darksky.net/forecast/YOUR DARK SKY KEY/44.9764016, -93.2486732?units=us&lang=en&r=0.436586280869
This is what I’mseeing.

togatown, 2019-01-13:
Pasting the URL into a browser and subbing in my API key works. I did notice you are using the default coordinates from the config file which leads me to believe that you are not saving your files properly after editing them unless you are coincidentally in the same exact location the config was setup for!

I can only assume from your posts that you are using the nano command line editor to alter the config and apikeys files like this: 

`nano ApiKeys.py` 

and

`nano Config.py`

I don't think you are saving the files after editing them. After making your edits, press Ctrl+o and then Enter to save the files. Then use Ctrl+x to exit the editor.

You could also use the File Browser on the Pi, the Folder Icon next to the Raspberry in the top left corner of the screen, to browse to home > pi > PiClock > Clock and double click on the ApiKeys.py and Config.py files to edit them in and GUI based editor instead of using the 'nano' command to edit them. This may be more comfortable to you as it will be more like Windows or OSx which you probably are more familiar with.

grayeagle57, 2019-01-13:
Where should I get the coordinates?

grayeagle57, 2019-01-13:
Found a website to get the coordinates. All is well. Thank you immensely for your help and guidance. Love this clock.

closed 2019-02-01 by n0bel