n0bel/PiClock #122: does not work with regional settings.

xenon462, 2019-01-09:
1. In the weather forecast text box, the units are displayed in inches. The system (si) is used in our region. Precipitation should be in centimeters. You need to change the ‘units’ parameter in the PyQtPiClock.py file line 406. probably better to put this parameter in the Config.py file and make it mutable (us, si, auto, ca). And you need to remove the unit conversion line 156-168.
[image]
2. Date and time are displayed with an error. I tried different locales, different ways nothing works. Probably a bug in Python2.7 itself.
I found a way to substitute the day of the week by the number of the day of the week.
        weekday = "{0:%w}".format(now)
        if (weekday == '0'):
            weekrus = u' Воскресенье '
        if (weekday == '1'):
            weekrus = u' Понедельник '
        if (weekday == '2'):
            weekrus = u' Вторник '
        if (weekday == '3'):
            weekrus = u' Cреда '
        if (weekday == '4'):
            weekrus = u' Четверг '
        if (weekday == '5'):
            weekrus = u' Пятница '
        if (weekday == '6'):
            weekrus = u' Суббота '
Для верхней строки это работает.
[image]
How can I do it in the same way for the date and time in the right column? I can not help.
3. I added data OZON, UV-index, visibility.

How to create a separate line with this data at the bottom of the page above the “phase of the moon” line?
here are my files
[Config.py.txt](https://github.com/n0bel/PiClock/files/2741846/Config.py.txt)
[PyQtPiClock.py.txt](https://github.com/n0bel/PiClock/files/2741848/PyQtPiClock.py.txt)

n0bel, 2019-01-12:
The date problem is some sort of UTF-8 issue.   I'll take a peek, hopefully others might try too.

BertLindeman, 2019-01-14:
It is not clear to me what is causing the error.
To test I took some parts of PyQtPiClock.py and made this script to see if this works:
```
# -*- coding: utf-8 -*-                 # NOQA

import sys
import os
import platform
import signal
import datetime
import time
import json
import locale
import random

def tick():

    global hourpixmap, minpixmap, secpixmap
    global hourpixmap2, minpixmap2, secpixmap2
    global lastmin, lastday, lasttimestr
    global clockrect
    global datex, datex2, datey2, pdy

    if Config.DateLocale != "":
        try:
            locale.setlocale(locale.LC_TIME, Config.DateLocale)
        except Exeption as e:
            print("locale.LC_time failed: %s" % Config.DateLocale)
            print(e)

    now = datetime.datetime.now()
    if Config.digital:
        timestr = Config.digitalformat.format(now)
        if Config.digitalformat.find("%I") > -1:
            if timestr[0] == '0':
                timestr = timestr[1:99]
        if lasttimestr != timestr:
            clockface.setText(timestr.lower())
        lasttimestr = timestr
    else:
        if now.minute != lastmin:
            lastmin = now.minute

    dy = "{0:%H:%M}".format(now)    #DATE  0:%I:%M %p------------------------
    print("dy=%s" % dy)
    if now.day != lastday:
        lastday = now.day
        # date
#        weekrus = ''
        weekday = "{0:%w}".format(now)
        if (weekday == '0'):
            weekrus = u' Воскресенье '
        if (weekday == '1'):
            weekrus = u' Понедельник '
        if (weekday == '2'):
            weekrus = u' Вторник '
        if (weekday == '3'):
            weekrus = u' Cреда '
        if (weekday == '4'):
            weekrus = u' Четверг '
        if (weekday == '5'):
            weekrus = u' Пятница '
        if (weekday == '6'):
            weekrus = u' Суббота '
        month = "{0:%m}".format(now)
        monthrus = ''
        if (month == '01'):
            monthrus = u'Января'
        if (month == '02'):
            monthrus = u'Февраля'
        if (month == '03'):
            monthrus = u'Марта'
        if (month == '04'):
            monthrus = u'Апреля'
        if (month == '05'):
            monthrus = u'Мая'
        if (month == '06'):
            monthrus = u'Июня'
        if (month == '07'):
            monthrus = u'Июля'
        if (month == '08'):
            monthrus = u'Августа'
        if (month == '09'):
            monthrus = u'Сентября'
        if (month == '10'):
            monthrus = u'Октября'
        if (month == '11'):
            monthrus = u'Ноября'
        if (month == '12'):
            monthrus = u'Декабря'

        # if Config.DateLocale != "":
        #     sup = ""
        ds = u" {1} {0:%d} {2} {0.year}г. ".format(now, weekrus, monthrus) #ДАТА
        #datex.setText(ds)
        #datex2.setText(ds)
        print("Update by xenon462:     %s \n" % ds)

def showFormattedDate(thislocale):
        now = datetime.datetime.now()
        Config.DateLocale = thislocale
        if Config.DateLocale != "":
            try:
                locale.setlocale(locale.LC_TIME, Config.DateLocale)
            except Exception as e:
                print("\tlocale.LC_time failed: %s" % Config.DateLocale)
                print("\t %s" % e)

        ds =  "{0:%A %d %B %Y}".format(now)
        print("Locale used %s : A d B Y %s " % (thislocale,ds))

        ds =  "{0:%A %B %d %Y}".format(now)
        print("Locale used %s : A B d Y %s " % (thislocale,ds))

        ds =  "{0:%H:%M}".format(now)
        print("Locale used %s : H M     %s " % (thislocale,ds))

        ds =  "{0:%I:%M %p}".format(now)
        print("Locale used %s : I M p   %s \n" % (thislocale,ds))

configname = 'Config'

if len(sys.argv) > 1:
    configname = sys.argv[1]

if not os.path.isfile(configname + ".py"):
    print "Config file not found %s" % configname + ".py"
    exit(1)

Config = __import__(configname)

lastmin = -1
lastday = -1
pdy = ""
lasttimestr = ""
weatherplayer = None
lastkeytime = 0
lastapiget = time.time()

tick()

showFormattedDate("ru_RU.UTF-8")
showFormattedDate("en_US.UTF-8")
showFormattedDate("nl_NL.UTF-8")

```
Installed  **ru_RU.UTF8** locale next to mine.

Current output looks good to me (just a native Dutch ;-) )
```
dy=21:32
Update by xenon462:       Понедельник  14 Января 2019г.

Locale used ru_RU.UTF-8 : A d B Y Понедельник 14 Январь 2019
Locale used ru_RU.UTF-8 : A B d Y Понедельник Январь 14 2019
Locale used ru_RU.UTF-8 : H M     21:32
Locale used ru_RU.UTF-8 : I M p   09:32

Locale used en_US.UTF-8 : A d B Y Monday 14 January 2019
Locale used en_US.UTF-8 : A B d Y Monday January 14 2019
Locale used en_US.UTF-8 : H M     21:32
Locale used en_US.UTF-8 : I M p   09:32 PM

Locale used nl_NL.UTF-8 : A d B Y maandag 14 januari 2019
Locale used nl_NL.UTF-8 : A B d Y maandag januari 14 2019
Locale used nl_NL.UTF-8 : H M     21:32
Locale used nl_NL.UTF-8 : I M p   09:32
```

Maybe for ease of configuring the prefered format for the date and time can be in the Config?
"{0:%A %d %B %Y}" for dayname day# month year
"{0:%A %B %d %Y}" for dayname monyh day# year

"{0:%H:%M}" for 24 hour clock format 
"{0:%I:%M %p}" for 12 hour clock format including AM/PM

In my Config I have 
```DateLocale = 'nl_NL.UTF-8' ```
and that works OK.
Except the date and time format ;-)

Hope this helps a bit.

BertLindeman, 2019-01-14:
The correct date string is produced by the formats in my previous post. Yet the display shows the garbage as shown in the first post in this issue.
Maybe the problem is that these calls are not UTF8 friendly?
```
        datex.setText(ds)
        datex2.setText(ds)
```

n0bel, 2019-01-15:
Yes this is what I suspected in my original comment.
I could (might) be a font problem too.
I've not tried any experiments yet to narrow down the issue.
Thanks for your tries.  A few less things for me to try.

BertLindeman, 2019-01-15:
Did a bit more fiddling and I do **not** understand, BUT it works at the moment for me.
@xenon462 could you try in the original PyQtPiClock.py
Add this line
```
        ds = ds.decode('utf-8')  # /BL 20190115 try to get Cyrillic date formatted without garbage
```
before these two lines:
```
        datex.setText(ds)
        datex2.setText(ds)
```
and see wether the large date on both "sides" of the clock is at least without garbage?
This does not yet help to translate the names of the days and the time format of the forcast boxes.
But - step by step ;-)

xenon462, 2019-01-16:
if you add the line 
ds = ds.decode ('utf-8')   then Log gives an error
File "PyQtPiClock.py", line 160, in tick
    ds = ds.decode('utf8')
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeEncodeError: 'ascii' codec can't encode characters in position 2-8: ordinal not in range(128)
[image]

BertLindeman, 2019-01-16:
Rats, my sd-card in my RPI2 just failed on me, so it may take me some time to rebuild it.

Thank you for reporting back.

Yesterday I ran with this "patch" using ru_RU.UTF8 in my Config without problem.
I saved a picture:
[image]

[image]

Just to be sure (from my backup before the crash) it looked like this:
```
    if now.day != lastday:
        lastday = now.day
        # date
        sup = 'th'
        if (now.day == 1 or now.day == 21 or now.day == 31):
            sup = 'st'
        if (now.day == 2 or now.day == 22):
            sup = 'nd'
        if (now.day == 3 or now.day == 23):
            sup = 'rd'
        if Config.DateLocale != "":
            sup = ""
        ds = "{0:%A %B} {0.day}<sup>{1}</sup> {0.year}".format(now, sup)
        ds = ds.decode('utf-8')  # /BL 20190115 try to get Cyrillic date formatted without garbage
        datex.setText(ds)
        datex2.setText(ds)

def tempfinished():
```

I ask, because the added line here is on line 123. While your error occurs on line 160.
:-) Just asking...

As said it may take me some time....

xenon462, 2019-01-16:
[image]

xenon462, 2019-01-16:
if you do not add the line ds = ds.decode ('utf-8') then the upper line works fine.
[image]

xenon462, 2019-01-16:
The problem with writing the day of the week on the right in the column.

xenon462, 2019-01-16:
I added the wrong line.
Now the top line is working.
I added ds = ds.decode ('utf-8') to the original PyQtPiClock.py file.
[image]
[image]

BertLindeman, 2019-01-16:
Ah, I see you are not using the original PyQtPiClock.py but your own updated one (with self-setting the weekday and month). 
I made the update on the original PyQtPiClock.py from github.
IF the update works on the original, then you would not need to update the script for this anymore.
We only need a way to put the preferred format somehow in the Config. And the same goes for the other date / time formats. I could make a pull request for it n a few days. (if @n0bel agrees off course).

Could I ask you to (temporary) use the **original** PyQtPiClock.py with my update to verify it does work?

xenon462, 2019-01-16:
I will use the original file for now. The date format suits me. Maybe it will be useful to others

xenon462, 2019-01-16:
I managed to fix the error of the days of the week.
I added    .decode ('utf-8')
[image]
in two places
[image]
[image]
thanks for the tip.

BertLindeman, 2019-01-18:
@n0bel Would you accept a pull request to place all date and time formats to Config.py (and Config-Example*.py) to "fix" this issue and #121 in one go.

Before I start on that I propose to add these variables:
``` 
# Non digital date and time formats
LtopDateformat = '{0:%A %B} {0.day}<sup>{1}</sup> {0.year}' # Locale full dayname and monthname, day of month number{th/rd} full year
Ltimeformat    = "{0:%I:%M %p}"                               # the default 12hour clock AM/PM = Time under the top-left weather conditions
Ltimesunformat = "{0:%I:%M %p}"                               # the default 12hour clock AM/PM = Time Sun-up/down
Ltimestampformat = '{0:%H:%M}'                                # 24-hour hour and minute = Timestamp of Rainview forcast
Lforcastdaytimeformat = '{0:%I:%M %p}'
Lforcastdayformat = "{0:%A}"
```

And to allow configurations like the one of @xenon462 
add this to the top of each Config-Example:
```
# -*- coding: utf-8 -*- 
  
"""   
     Configuration module for PyQt clock 
"""     
```

Updates for PyQtPiClock.py would be 

- 3 updates to add .decode('utf-8')
- add reading the Config for the new date / time variables
   As the number of variables is rather high I would offer a test_formats.py to verify a Config.py
- replace the date / time formats  in PyQtPiClock.py with Config.variables.

As example part of test_formats.py similar to what I intend to add to the clock:
```
# define default values for new/optional config variables.

try:
    Config.LtopDateformat
except AttributeError:
    Config.LtopDateformat = "{0:%A %B} {0.day}<sup>{1}</sup> {0.year}"
    print("Warn: take default Config.LtopDateformat:        \t'%s'" % Config.LtopDateformat)
```

yura-n-d, 2019-05-28:
Yourproject is rather nice. Thank you! But while repeating it I faced two problems.

1. At your printscreen in the right colomn days of week go not in the regular order: Wednesday, Wednesday, Thursday, Wednesday, Thursday, Friday, Saturday, Sunday, Monday.

I've got the same wrong sequence and I would like to correct it.

2. How can I display pressure in "mm Hg"?

xenon462, 2019-05-28:
1. in the right column of the first, second and third line above the weather forecast for 3 hours.
the fourth line is the forecast for today for the whole day.
in the fifth, sixth, seventh, eighth, ninth line weather forecast for the following days of the week.
2. to display the pressure in mmHg it is necessary to add after the line 256 in the PyQtPiClock.py file
line with this content:  pressmmrtst = int(f['pressure']) * 0.750064#  Давление в мм рт ст
and line 257, correct for a line with the following content:  press.setText(Config.LPressure + str(int(pressmmrtst)) +u' мм рт ст  ')

yura-n-d, 2019-05-28:
More thanks, xenon462!
Вечером попробую. Поделитесь мудростью, как вам фазы луны удалось кириллицей отобразить? В этой теме не нашел, если меняю на кириллицу - программа не запускается с ошибкой. Спасибо!

yura-n-d, 2019-05-28:
More thanks, xenon462!

xenon462, 2019-05-28:
My files to display the phase of the moon in Cyrillic.
[image]

[image]

[Config.py.txt](https://github.com/n0bel/PiClock/files/3227599/Config.py.txt)
[PyQtPiClock.py.txt](https://github.com/n0bel/PiClock/files/3227604/PyQtPiClock.py.txt)

yura-n-d, 2019-05-30:
Xenon462, thank you very much! Everything is ok now.
[image]

closed 2020-12-24 by n0bel