n0bel/PiClock #124: Weather forecast units is hardcoded 'us'

BertLindeman, 2019-01-18:


BertLindeman, 2019-01-18:
There are conversions in PyQtPiClock that produce curious results if I use auto for the units ;-)
Currently the temperature in about minus 1.3 Celsius and "converted from Fahrenheit into Celsius"
So I was thinking a bit too easy about this. Picture with the wrong values ;-)

[image]

Minus 20 Celsius is here very rare, for @xenon462 it is more common ;-)

xenon462, 2019-01-18:
I removed the multiplication by the coefficient. Everything is working fine for me.
[image]

BertLindeman, 2019-01-18:
Ah, so not so difficult, but to generalize it sti;; takes some programming the conversion depending  on the units variable. Thank you for the update.

closed 2020-12-24 by n0bel