n0bel/PiClock #127: Background image change in new PiClock

feh123, 2019-01-22:
Tried to alter default background image with my own jpg or png - no success. Just see blank, black background. Tested jpg's with EXIF data cleared, with dimensions (both px's and file size) similar to default pictures, No clear error as PiClock loads.

BertLindeman, 2019-01-22:
Ok,
First let me link to issue #125 as we came from there.
In #125 I noted that missing background file does not report any error.

Let me paste your image back to you - seems fair, right?
[image]

There is NO error reported on my host for this image.

Next is a python script to test your Config.py
Rename the file after download from `*.py.txt` to `*.py` and run in your `/home/pi/PiClock/Clock` directory
 
[test_configuration.py.txt](https://github.com/n0bel/PiClock/files/2784004/test_configuration.py.txt)

feh123, 2019-01-22:
Yes - as you suggested - sending/receiving the jpg over Github certainly does something - the file of the jpg you sent me works!

Not checked the .py yet. Thanks for the help.

BertLindeman, 2019-01-22:
Oke  . . . .

feh123, 2019-01-23:
Hi I changed the test file name to a py file in Notepad++ and transferred it to the Clock folder with FileZilla. I opened the file via VNC with Thonny. My config.py contains the default setup apart from the background image - which is one of the many jpgs I tried that do not work. PiClock is working without a background.

I pressed the run icon in Thonny and got the Shell message that I cannot copy/paste unfortunately. So in summary it says: 
Python 3.5.3 (/user/bin/python3), >>>%Run file, Traceback (most recent call last): file test_configuration.py line 64 print "config file....," Syntax error missing parentheses in call 'print'. >>>

Am I doing something wrong? Is there something different between our two OS setups?

BertLindeman, 2019-01-23:
The current PiClock uses **python2** 
The failure you describe is running **python3** while PiClock  needs python**2**.
The same goes for the ```test_configuration.py```.

You can run the test_configuration program in a console or terminal session in the 
 ```/home/pi/PiClock/Clock``` directory like this: 
```python2  test_configuration.py Config```
Was that the doc you missed?

feh123, 2019-01-23:
Hi using the remote terminal app on my Windows 10 PC. I can get the program to work. How can I copy and send you the output? 
Not sure what doc you mean!
Thanks.

BertLindeman, 2019-01-23:
:-) Correct, it's not documented, as it is only meant to verify the Config.py contents. 

By all means paste the output.

Main question is does it report your background image as valid, like so:
```
Config.background        images/clockbackground-springflowers.jpg Exists:True
```

To verify the file is an image you could also issue in the remote terminal app:
```
file images/clockbackground-springflowers.jpg
```
With YOUR image filename of course ;-)

feh123, 2019-01-23:
So from a long list of true it picks out as false: background, squares1, squares2, Icon hail and Icon tornado. However file image check command returns a positive and identifies my background jpg. It seems confusing to me!

BertLindeman, 2019-01-23:
Maybe better?
Show the background line from your config and an
```
ls  -l 
```
from that background file

feh123, 2019-01-23:
In PiCLock/Clock/images I use ls -l and see all the expected images including squares and my background image. The icon files are in PiClock/Clock.

BertLindeman, 2019-01-23:
Could you please copy / paste the ls -l output 
only the relevant lines is OK (include the top line). 
and the line of Config.py that describes the background image.
Like so (between 3 backtics so it's well formatted):
```
pi@mammoet:~/PiClock-dev/PiClock/Clock$ ls -l images/
-rw-rw-r-- 1 pi pi  538685 Jan 22 14:05 berlin-at-night-mrwallpaper.jpg
```
and

```
background = 'images/berlin-at-night-mrwallpaper.jpg'
```

From what you write here there is nothing wrong but still is does not work.

I can better help if I can see it.

feh123, 2019-01-23:
Okay:
ls -l                          
total 18740                                                                            
-rw-r--r-- 1 pi pi  729939 Jan 22 09:03 abstract-arctic-art-3262401                    
.jpg                                                                                   
-rw-r--r-- 1 pi pi  729940 Jan 22 09:03 abstract-arctic-art-3262402                    
.jpg                                                                                   
-rw-r--r-- 1 pi pi  755487 Jan 22 09:03 abstract-arctic-art-3262403                    
.jpg                                                                                   
-rw-r--r-- 1 pi pi  637528 Jan 22 09:03 abstract-arctic-art-3262404                    
.jpg                                                                                   
-rw-r--r-- 1 pi pi  538685 Jan  7 12:56 berlin-at-night-mrwallpaper                    
.jpg                                                                                   
-rw-r--r-- 1 pi pi  525022 Jan  7 12:56 clockbackground-chris.png                      
-rw-r--r-- 1 pi pi 1454101 Jan  7 12:56 clockbackground-jean.png                       
-rw-r--r-- 1 pi pi 1132004 Jan  7 12:56 clockbackground-kelly.png                      
-rw-r--r-- 1 pi pi 1235907 Jan  7 12:56 clockbackground-kevin.png                      
-rw-r--r-- 1 pi pi  984012 Jan  7 12:56 clockbackground.png                            
-rw-r--r-- 1 pi pi  143373 Jan  7 12:56 clockface3-darkblue.png                        
-rw-r--r-- 1 pi pi  133647 Jan  7 12:56 clockface3-darkgreen.png                       
-rw-r--r-- 1 pi pi  158026 Jan  7 12:56 clockface3.png                                 
-rw-r--r-- 1 pi pi 1126281 Jan 22 18:19 coldsnow.jpg                                   
-rw-r--r-- 1 pi pi  306840 Jan 22 10:07 cold-snow-white-6056156A.pn                    
g                                                                                      
-rwxrwxrwx 1 pi pi 1159395 Jan 22 09:59 cold-snow-white-6056156.png                    
-rwxrwxrwx 1 pi pi 1734192 Jan 22 11:33 cold-snow-white-6056157.jpg                    
-rw-r--r-- 1 pi pi 1309292 Jan 22 13:06 cold-snow-white-6056158.jpg ****                   
-rwxrwxrwx 1 pi pi 1309292 Jan 22 12:45 coldsnowwhite6056158.jpg                       
-rwxrwxrwx 1 pi pi 1126281 Jan 22 09:54 cold-snow-white-605615.jpg                     
-rwxrwxrwx 1 pi pi 1667468 Jan 22 11:39 cold-snow-white-60561.jpg   

The background I am trying to use is ****.

From Config.py file: 
background = 'images/cold-snow.white-6056158.jpg'                                      
squares1 = 'images/squares1.png'                                                       
squares2 = 'images/squares2.png'                                                       
icons = 'icons-lightblue'                                                              
textcolor = '#bef'                                                                     
textfont = 'arial'                                                                     
clockface = 'images/clockface3.png'                                                    
hourhand = 'images/hourhand.png'                                                       
minhand = 'images/minhand.png'                                                         
sechand = 'images/sechand.png'  

Added a few more lines for interest. I have seen an error - the . in cold-snow.white should be an -! I will check this but send the file you now.

BertLindeman, 2019-01-23:
So I think you found the culprit:
```
background = 'images/cold-snow.white-6056158.jpg'
```
The character before `white` 
should it be a dash `-` and now is a period '.'

feh123, 2019-01-23:
Yes - I thought I had checked this 1000x. All working now. I will close the issue. Thanks for all your help. Much appreciated.

closed 2019-01-23 by feh123

BertLindeman, 2019-01-23:
Glad you got it worked out.
anytime..