n0bel/PiClock #129: New install corrupting file system

cyptex16403, 2019-01-22:
On an new PI 3 B, fresh PiClock install with base config, very slow to launch and upon exit file system is corrupted and having to run fsck to recover.  Not sure what could be causing this?  Any Suggestions?

billskeen68, 2019-01-23:
I updated to the new PiClock2 and now having the same issue.

n0bel, 2019-01-23:
PiClock2 is not updated to the replacement weather and maps apis.   You need to stick with PiClock, not PiClock2.   As far as things being slow, and the other error you're getting, I'd save your configs to some other computer, then start from scratch with a new SD card.

billskeen68, 2019-01-23:
Apologies, I updated my comment. I meant the new version using DarkSky.

n0bel, 2019-01-23:
> 
> 
> I updated to the new PiClock using DarkSky and now having the same issue.
> 
> Also getting this error when ssh in:
> 
> /usr/bin/xauth: error in locking authority file /home/pi/.Xauthority
> 
> Read somewhere it might have to do with being root in pi home directory...looking into setup to see if I missed a step.

This can happen if you run some things as root.   If you're quite familiar with linux, or search alot around the internet you may be able to fix it.    But from my perspective, I'd just start with a new SD card, because that could be the issue anyway.

But really, root should not have been required to update.

cyptex16403, 2019-01-24:
The new SC recommendations did the trick....thank you all!  It must have been defective from the start...even though it was new as well....no matter....back running as expected....

closed 2019-02-01 by n0bel