n0bel/PiClock #131: Radar shows snow as rain?

ScottChapman, 2019-02-01:
So... It's winter. It's super cold. Radar shows a lot of green moving in. But it is snow not rain.

Shouldn't it be a different color?

feh123, 2019-02-01:
Mine is similar. I have never seen anything other than rain colours even though snow has been a regular feature here in Cambridge. I have an android app with weather radar from WetterOnline Meteorologische in Bonn correctly showing the snowfall here (now) and across Europe.
But I think this is normal for RainView - the live radar weather map on their website has a colour legend - it does not include snow. So the written forecast says snow but the radar shows only rain colours and the icon is not a snow icon rather the rain one - do we have icons differentiate that rain and snow?

BertLindeman, 2019-02-01:
We do have snow  icons. And I have seen it being used. The icon field in the DarkSky json data determines which icon is shown. (Assuming DarkSky api is used). I also see only "rain" colors.

BertLindeman, 2019-02-01:
The color legend  for rainview can be customized. In PiClock at the moment we use "color=3"
```
https://tilecache.rainviewer.com/v2/radar/1549009800//256/5/17/11.png?color=3
```
From [rainview api doc](https://www.rainviewer.com/api.html) 

Color Scheme Identifier | Name
-- | --
0 | Original (default)
1 | Universal Blue
2 | TITAN
3 | The Weather Channel
5 | NEXRAD Level III
6 | Rainbow @ SELEX-IS

An example can be seen by going to [https://www.rainviewer.com/](https://www.rainviewer.com/)
on the righthand top there is a button "options". 
Change the color scheme and instantly see it change.
I like the "0 = Original" color scheme to best fit how I would interpret the colors.

feh123, 2019-02-01:
Thanks for the info and apologies for all the questions but how do these things fit together? I am using the DarkSky api - I see a DarkSky logo top left on my PiClock. The radar images are labelled RainView.com - rainviewer.com (a typo I guess?). So the DarkSky api should have used a snow icon this morning? And DarkSky use RainViewer for the weather radar? Can RainViewer show snow?

BertLindeman, 2019-02-01:
Ho, a lot of questions ;-) 
As I understand it:
DarkSky produces the current weather and forecasts and those are used for:
- the top description and icon
- the boxes on the right hand side.
rainviewer.com is used to create the animation of rain (and other precipitation (if that is written like this) )
I did not find the possibility to show snow there. 
But maybe someone else might ;-)

BertLindeman, 2019-02-01:
To see what data was produced by DarkSky you can look in your PiClock log file and copy a line like this:
```
https://api.darksky.net/forecast/00be092dcf9c308ad93923438604d906/51.892625,4.627072?units=us&lang=nl&r=0.871422316988
```
Paste it in a webbrowser and you'll see a json file. 
In there are sub fields `icon` that value is used in PiClock.

feh123, 2019-02-01:
Hi - yes my log shows that DarkSky used the icon snow - I will look out for it more closely now - more snow is predicted here.

So access to RainViewer does come from DarkSky? 

I could not find snow on the RainViewer either but there may be a way to get it. If anyone is looking (and good at searching) - if you could add clouds that would be good too!

BertLindeman, 2019-02-01:
> So access to RainViewer does come from DarkSky?

rainviewer is called from PiClock itself:

```
def getTiles(self, t, i=0):
        t = int(t / 600)*600
        self.getTime = t
        self.getIndex = i
        if i == 0:
            self.tileurls = []
            self.tileQimages = []
            for tt in self.tiletails:
                tileurl = "https://tilecache.rainviewer.com/v2/radar/%d/%s" \
                    % (t, tt)
                self.tileurls.append(tileurl)
        print self.myname + " " + str(self.getIndex) + " " + self.tileurls[i]
        self.tilereq = QNetworkRequest(QUrl(self.tileurls[i]))
        self.tilereply = manager.get(self.tilereq)
        QtCore.QObject.connect(self.tilereply, QtCore.SIGNAL(
                "finished()"), self.getTilesReply)
```

ScottChapman, 2019-02-01:
So, looking at the rainviewer site, it looks like they do not make a distinction between precipitation types:
[image]
Thats using satellite background with weather channel color scheme (which is what I assume we use). But if I user Wunderground (weather channel) Wundermap:
[image]

feh123, 2019-02-01:
Ah, thanks for the explanation. Does this mean we could change the tile url and get snow (or clouds?). I guess not!

n0bel, 2019-02-01:
As has already been said, RainView does not change the color based on frozen, mixed or not.

WU did this by looking at the temperatures (or some other  method) in the area of the pixel being drawn.   So to emulate WU, you couldn't just switch the whole map to blue (type 0 or 1 for RainView), you'd have to do it by some sort of temperature map.

We do have two methods to control the map colors.
1) Choose a different color scheme based on current temperature.
2) Use the color scheme 'source' which is dBZ values, and then color them in the PiClock clode (by overlaying some kind of temperature map) [ anyidea where to get a temperature map? OpenWeatherMap does have a temperature map (and precipitation map for that matter) but they guarentee updating as "less than 3 hours", which I would find practically useless ]

I'm really up for #2, if we can find a temperature map data source.

Kevin

ScottChapman, 2019-02-01:
Wow, that sounds legitimately like a pain in the butt.

BertLindeman, 2019-02-01:
> Wow, that sounds legitimately like a pain in the butt.

**have** to respond with :-)

So we have to find the temperature of each pixel of the radar squares and  determine what color the precipitation tilepixel should have. 
Wow.

BertLindeman, 2019-02-02:
Wetteronline.de does not disclose their offerings for api usage. You need to contact them to get a quote. 
So I assume no free developer rate. 
Did not see (at a glance) snow radar at [WetterOnline API pdf](https://st.wetteronline.de/mdr/p_marketing/1.11.238/media/weather-api/WetterOnline_Weather-API.pdf) and [WetterOnline API interface pdf](https://st.wetteronline.de/mdr/p_marketing/1.11.238/media/weather-api/WetterOnline_Weather-API_interface_description.pdf) 
Good thing for them: 15 minutes updates.

BertLindeman, 2019-02-02:
### MeteoBlue 

Weather map (layer) | Price per year
-- | --
Temperature | 400 EUR
Precipitation | 800 EUR
Snowfall | 800 EUR

They deliver also snow precipitation, but this pricing is way too high for me.

###Weatherbit.io

Offers a non-commercial plan: [weatherbit.io pricing](https://www.weatherbit.io/pricing)
But cannot find radar images in their api.

### AccuWeather
6 Month trial. Each developer is limited to 500 calls per day.
The radar images are NOT included in the [trial](https://developer.accuweather.com/pricing-and-data-packages)

No success so far...

ScottChapman, 2019-02-04:
Just FYI - I took the liberty of reaching out to the developer of `RainView` and he was super interested in this project (plans on setting one up on his pizero!).

Anyway he is working on it using the approach by @n0bel using the temp map from OpenWeatherMap (updated every 15-20 minutes) which seems totally reasonable.

Hopefully he'll keep us posted on it (and hopefully it can get implemented before spring!)

n0bel, 2019-02-07:
OpenWeatherMap guarantees temperature map updates as "< 3 hours", at the free tier level.   I've been meaning to run some pulls and compares to see how often it "actually" ("usually") updates.

@ScottChapman, thanks for contacting Mr. RainView.

merky1, 2019-02-08:
Maybe look at the aeris weather map layers?  
https://www.aerisweather.com/support/docs/aeris-maps/reference/map-layers/#all:radsat

Downside would be yet another api to sign up for.

ScottChapman, 2019-02-28:
FYI - I noticed this morning that my radar showed nice shades of white/blue while it was snowing:
https://www.rainviewer.com/api.html

(and there was much rejoicing)

closed 2019-02-28 by ScottChapman