n0bel/PiClock #134: NOAA radio stream

GrampDamnpus, 2019-02-02:
I cannot get the NOAA stream to work. My url ( https://www.weather.gov/media/mfl/nwr/MIAZFPWPB.mp3 ) works fine on my windows machine. I can add other urls, for say a generic ringtone or whatever, and those play fine when I press f2. When I try and play my weather URL I get the error: [mpg123.c:629] error: Cannot open https://www.weather.gov/media/mfl/nwr/MIAZFPAPF.mp3: File access error. (code 22)

In the PiClock log I get the same error. Im not sure exactly what Im doing wrong.

BertLindeman, 2019-02-03:
Maybe a longshot, but the manpage of mpg123 states:
```
any name starting with ``http://'' is recognized as URL 
```
I do not have the noastream working on my clock, but I **do** see a difference between the URL starting with `https:` and `http:`.

```
[src/libmpg123/readers.c:1184] error: Cannot open file https://www.weather.gov/media/mfl/nwr/MIAZFPWPB.mp3: No such file or directory
main: [src/mpg123.c:708] error: Cannot open https://www.weather.gov/media/mfl/nwr/MIAZFPWPB.mp3: File access error. (code 22)
```
or
```
HTTP request failed: 400 Bad Request
main: [src/mpg123.c:685] error: Access to http resource http://www.weather.gov/media/mfl/nwr/MIAZFPWPB.mp3 failed.
```
Did not see a confirmation on the web (yet).
As I said no audio on my Clock....

BertLindeman, 2019-02-03:
Have installed mpg123 on my rpi2 and tested with a local mp3: works
Next test as user pi (did not need to do this as root):
```
mpg123 -v http://radio.weatherusa.net/NWR/KIH54.mp3
```
This works for me.

Then:
```
mpg123 -v https://www.weather.gov/media/mfl/nwr/MIAZFPAPF.mp3
```
No success, so I blame the https on mpg123:
```
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes
Decoder: generic
Trying output module: alsa, device: <nil>
[src/libmpg123/readers.c:1182] error: Cannot open file https://www.weather.gov/media/mfl/nwr/MIAZFPAPF.mp3: No such file or directory
main: [src/mpg123.c:732] error: Cannot open https://www.weather.gov/media/mfl/nwr/MIAZFPAPF.mp3: File access error. (code 22)
```

Tried to find another webbased weather podcast using https but could not find one.

BertLindeman, 2019-02-03:
Ah ;-) Looked at the page source of `https://noaaweatherradio.org/` and looked for 
`data-path`
And found some **https** URLs:  
- `https://noaaweatherradio.org/Hurricane.mp3`
- `https://www.weather.gov/media/mfl/nwr/MIAZFPBLG.mp3`
- `https://noaaweatherradio.org/content/NHCoffseason.mp3`
- `https://www.weather.gov/media/mfl/nwr/MIAZFPBLG.mp3`
- `https://icy3.boisestatepublicradio.org:9016/stream`

There are more, but I found this conclusive enough that all these **https** streams fail,
and are not supported by mpg123.
Annoys me that I can not find doc for mpg123 stating that https streams are not supported.

Chose another http (no S) stream from Florida and that one does work:
```
mpg123 -v http://wxradio.dyndns.org:8000/Tampa-FL-KHB32.mp3
```
I ran all these tests in a terminal session as user **pi** so there was no need to update the PiClock Config and stop/start it so often.

GrampDamnpus, 2019-02-03:
Thanks you so much for taking the time to help sort this out, its been killing me. I feel dumb that I didnt even think about https being an issue. Sadly Tampa wont work as its the opposite side of the state so the noaa feed wouldnt be all that helpful. Ill look through them all again and see what I can do.

BertLindeman, 2019-02-03:
No problem.
Now it's up to you to find a better station ;-)
Success

BertLindeman, 2019-02-03:
Hope you do **not** need a severe hurricane warning.
Very good you dropped the request for https support on sourceforge!

GrampDamnpus, 2019-02-04:
One of the devs of mpg123 got back to me and supplied me with a command that works to play the needed stream from the shell:
curl https://stream.url | mpg123 -

Ive spent a few minutes digging through PyQtPiClock.py to see a way to add that command to the player call, sadly this is WAY above my pay grade as it were.

BertLindeman, 2019-02-05:
First of all you save the current PyQtPiClock.py as backup.

I tried this command in a terminal before trying to change the clock:
```
curl --silent https://www.weather.gov/media/mfl/nwr/MIAZFPAPF.mp3 | mpg123 -q -
``` 
This worked for me. (do not forget the trailing dash)

You found the location in the clock, I assume you know how to edit PyQtPiClock.py
(e.g. in a terminal session as user pi ```nano /home/pi/PiClock/Clock/PyQtPiClock.py``` 
or another editor that you like)

Update  PyQtClock.py the line of code you indicated at or near line 828:
**Change**
```
                            ["mpg123", "-q", Config.noaastream])
```
**into** 
``` #  20190205 NOAA radio stream #134
#                            ["mpg123", "-q", Config.noaastream])
                            ["curl --silent ", Config.noaastream, "| mpg123 -q - "])
```

I made a comment of the original line so it is easier to fallback if need be.
Warning: I do not have buttons on my clock, so I cannot verify that this really works.
No error was reported though which only proves there are no syntax errors....

GrampDamnpus, 2019-02-11:
You sir are a champion, I had gotten as far as the curl command `curl https://stream.url | mpg123 -` but for the life of me I couldnt figure a way to implement it. The fella that does mpg123 suggested a script, and pointed me in the proper direction there I was just lost on how to actually do it. Now to test your solution and see.

BertLindeman, 2019-02-11:
Grumble... ;-) (No buttons here to set the stream on . . . So cannot really test.

Thinking about a PATH difference between the terminal and within python.
Would you issue in the terminal ```which curl``` 
That reports the path where curl is located.
In my system it's at:
```/usr/bin/curl```

And then change the mpg123 line to include the path to curl.
So like:
```
                            ["/usr/bin/curl --silent ", Config.noaastream, "| mpg123 -q - "])
```
And retest?
Would be a temporary circumvention, as I do not know exactly how/where the PATH is set in python.

GrampDamnpus, 2019-02-11:
Mine is the same `/usr/bin/curl` 
Edited and testing now.

GrampDamnpus, 2019-02-11:
Same again. 

```Traceback (most recent call last):
  File "PyQtPiClock.py", line 829, in keyPressEvent
    ["/usr/bin/curl --silent ", Config.noaastream, "| mpg123 -q - "])
  File "/usr/lib/python2.7/subprocess.py", line 390, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1024, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory

```

BertLindeman, 2019-02-11:
We use subprocess.Popen and that is more demanding than I thought.
At the moment I can test running ssh on my laptop =-> rpi2 where the clock resides.
And heee, on my laptop I **do** have a F2 key ;-) and then going to another room (to rpi2) plug-in headphone. Awell...

At the moment I have changed the line into:
```
                            ["curl --silent " + str(Config.noaastream) + " | mpg123 -q -"], shell=True)
```
So the command is converted into one complete string and we state that it should be processed as shell command.
and that works here...
Curious though.

GrampDamnpus, 2019-02-11:
I had been reading online and saw the shell=true command, just didn't have a clue how to implement it. I'm not home ATM but when I get there I will try it. And thanks again for all your time and help in this. As I learn more I'll to do my best to help others as much as I can.

GrampDamnpus, 2019-02-11:
Ok I got home and tested the current line: `["curl --silent " + str(Config.noaastream) + " | mpg123 -q -"], shell=True)` 
It starts great, but if you hit f2 to stop the stream it continues to play, and if you press it a thrid time it plays a second overlapping stream :) 

Getting there.

BertLindeman, 2019-02-11:
Did you have this strange F2 behaviour "using a noaastream with http (no https)" 
on the original PyQtClock?

I am trying to findout if **this set of mpg123 updates** is the cause of this strange F2 behaviour.

Or: any symptoms in the log related to the stream / keys?

n0bel, 2019-02-11:
try
```
sudo apt-get update
sudo apt-get install mpv
```
Go back to the original code and replace mpg123 with mpv.
mpv is slower to start.. so it takes several seconds for the stream to start.

n0bel, 2019-02-11:
> It starts great, but if you hit f2 to stop the stream it continues to play, and if you press it a thrid time it plays a second overlapping stream :)

piclock doesn't know how to kill all the subprocesses.

GrampDamnpus, 2019-02-11:
> Did you have this strange F2 behaviour "using a noaastream with http (no https)"
> on the original PyQtClock?
> 
> I am trying to findout if **this set of mpg123 updates** is the cause of this strange F2 behaviour.
> 
> Or: any symptoms in the log related to the stream / keys?

No, other than the https it worked fine with http streams.  

n0bel, I assume that you mean the OG code we commented out. So I'm testing that now with MPV like you suggested.

GrampDamnpus, 2019-02-11:
First try I got `Error parsing option q (option not found)
Setting command line option '--q=' failed.`

I removed the ` "-q",` and retested. It played a little over a minute/minute and a half after pressing F2, and then stopped when F2 was pressed a second time.

BertLindeman, 2019-02-12:
Noticed that the audio stream process stays active if the clock has been stopped.
I used `F4` to stop.
In `htop` I see that mpv keeps using cpu.

GrampDamnpus, 2019-02-12:
mpv works sorry if. The delay from press to play is not really optimal. At least on my setup.

closed 2020-12-24 by n0bel