n0bel/PiClock #137: Radar animation tick speed

GrampDamnpus, 2019-02-05:
Is there a way to slow down the radar frames? Currently it plays all the radar frames in succession in just over a second. I figured out you can add more individual frames by increasing line 458, `self.anim = `. But that doesn't slow down their tick rate, just adds more frames further into the history. It would be nice if each frame stayed on screen for a second making the entire radar sequence a little longer.

n0bel, 2019-02-07:
@GrampDamnpus 
In PyQtPiClock.py:
```
    def wxstart(self):
        print "wxstart for " + self.myname
        self.timer.start(200)
```
the 200 represents 200 milliseconds between radar frames. 

The time between each frame set is 5 times larger.  (The 5 in the following code snip from elsewhere in the same file)
```
        if self.displayedFrame == 0:
            self.ticker += 1
            if self.ticker < 5:
                return
```

Hope this helps.

merky1, 2019-02-08:
I found that setting the radar frames to 1000 “felt” right.

GrampDamnpus, 2019-02-11:
Awesome, Thats exactly what I was looking for.

closed 2019-02-11 by GrampDamnpus