n0bel/PiClock #139: Weekly Forecasts and Radar does not update

jdkitzerow, 2019-02-07:
I am having trouble with the weekly forecasts and radars updating.  I literally need to shutdown and restart the Pi a couple times before it updates to the current day.  The time on the top middle is correct, just not the weather information until I restart it a couple times.

I even crontab to reboot the pi every day, but its still not updating

Any suggestions?

maserowik, 2019-02-07:
Do not want to hijack this thread but I have been having the same issue.  

I am running PiClock on a Rapsberry pi zero w.

 

I went back and reload my image 

I  just the Piclock and did not configure any other feature like the led, remote control or the weather probes.

 

So far all has been running without the need to shutdown.

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

jdkitzerow, 2019-02-08:
Hijacked!!! Nah, just kidding... Here is a picture of how its Today's date with yesterdays weather....
[image]

n0bel, 2019-02-10:
These issues seems like some kind of caching issue.  As if there's a caching proxy between you and the provider.   This might be of help to determine that:   http://www.lagado.com/tools/cache-test

closed 2019-03-07 by n0bel