n0bel/PiClock #144: Weather Opacity

FrissonFurness, 2019-03-08:
Hi. I am extremely new to coding and using a Pi but I decided to take this on as my first project. I've managed to retrieve all the data correctly and i'm finding it very nice to use. I only have one issue with it, and it's changing the opacity of the "DarkSky" weather forecast displayed on the windows on the bottom left. I have seen that you can add different layers and have the forecast sit between the base and top layer but I wish to just change the opacity of it. Any help would be really appreciated.

Thanks :)

n0bel, 2019-03-13:
I'm not understanding what you need.
There is no forecast data on the bottom left.
As far as opacity goes, if its a text/label box (QLabel), then add   
```
 opacity: 200;  
```
to the style settings of the label  (find or create a setStyleSheet for the QLabel.

JayHauss, 2019-05-13:
I too have a question regarding weather opacity that I think applies to this issue. 

If you go to https://www.rainviewer.com/ they provide an option for "Radar Layer Opacity", with a default value of 83.

The PiClock seems to be using this default '83' value and it completely hides the underlying map. I can't find where you would specify a different opacity value in the PyQtPiClock.py file. Ideally I'd like it to be something like '35'. 

Right now if there's even the slightest cloud coverage, the radar tile just looks like a blotch of multicolored 8-bit squares and you can't see the underlying map at all.

SerBrynden, 2019-05-19:
@JayHauss I have a [fork of PiClock](https://github.com/SerBrynden/PiClock) which uses customized (dark) map layers with a bottom map layer of land and water only, then the weather radar layer, and then a top transparent layer of labels, roads, borders, and markers only. The map layers were customized using the free [Mapbox Studio](https://www.mapbox.com/mapbox-studio/).  This was done so that rain clouds in the radar don't obscure the map, thus providing reference as to where the rain is located.

closed 2020-12-24 by n0bel