n0bel/PiClock #146: Reduce Number of Calls to Dark Sky

Impchucker, 2019-03-12:
I changed over to Dark Sky two days ago, and I notice that I'm calling Dark Sky 48 times per day. At that rate, I'll easily exceed the 1,000 free calls per month. I'm trying to keep it free, so I would like to drop that to 32 calls per day. I'm out of practice coding this thing so I would appreciate any help y'all could give in achieving that. Or, is this something that will ruin my weather feed and I need to suck it up and pay?

BertLindeman, 2019-03-12:
I think you are incorrect about "1000 free calls per month"
From [DarkSky FAQ web page](https://darksky.net/dev/docs/faq)
``` 
How much does it cost to use the Dark Sky Forecast API?

You get 1,000 API calls for free each day. 
After that, we prorate each additional call at a rate of $0.0001 per call. 
The counter resets every day at midnight UTC. 
As long as you make fewer than 1,000 calls each day, the API is free.
```

If you still want to lower the weather update frequency, it's in your Config.py:
I am  using
 ```weather_refresh = 45    # minutes```

Impchucker, 2019-03-12:
Wow! I should have read more carefully. You are totally right. The dangers of reading before coffee are real. Thank you!

closed 2019-03-12 by Impchucker