n0bel/PiClock #151: Local temp sensor not showing up

xxxDrewedxxx, 2019-03-20:
I have a DS18B20 sensor that I can have the Pi see using:
cd /sys/bus/w1/devices/
ls

However the PiClock is not displaying it.  I deleted my old version of the clock ( actually formatted the SD card!) and started from scratch.  Any other ideas on where I can check?

BertLindeman, 2019-03-20:
Just two quick idea's (it's late here)

1. Did you adapt `/home/pi/PiClock/Temperature/TempNames.py` to fit the id of your sensor?
2. Does the TempServer run?

xxxDrewedxxx, 2019-03-21:
1. Yes I have
2. How do I check that?

togatown, 2019-03-21:
There are a couple of ways to check if TempServer.py is running. 

Open your browser and go to http://**Your_Pi_IP_Address**:48213/temp. You should get a json response which looks something like this:
`{"temps": {"temp-01142f9c611d": "74.9"}, "temp": "74.9"}`

Another method is to SSH into your Pi and enter the 'htop' command. Press F4 to filter your returns and type tempserver (htop is not case sensitive). You should see it in the list.

If it's not running, you can attempt to start it manual from a terminal to see if an error is returned. 
`cd PiClock`
`cd Temperature`
`python TempServer.py`

xxxDrewedxxx, 2019-03-21:
Thanks for the help.  It is working now.  I don't know what was the cause, but honestly, I'm thinking my bread board might be to blame.

togatown, 2019-03-22:
If you have an old computer, scavenge a ribbon cable from it and splice it to your sensor wires. Use heat shrink tubing to secure the connections after soldering.

xxxDrewedxxx, 2019-03-22:
Ok, I have everything up and running well.   I would like the data from multiple sensors to display on separate lines.   Is this possible?

togatown, 2019-03-23:
Yes but you may have to **move to QLabel named 'temp'** from its default position, its top position will have to go up or the text will run into the Sunrise/Set/MoonPhase line. The line you are concerned with is:

`temp.setGeometry(0, height - 100, width, 50)`
The values are as follows (x, y, width, height). X being the left of the element and y being its top. 

You can ignore the left and width values, you are concerned with the top and height. Try adding 50 for each line and tweak as needed. So for 1 additional sensor:
`temp.setGeometry(0, height - 150, width, 100)`

for two additional sensors:
`temp.setGeometry(0, height - 200, width, 150)`

You will also need to **alter the function 'tempfinished'** to add a newline character after each temp to get them on seperate lines.
```
def tempfinished():
    global tempreply, temp
    if tempreply.error() != QNetworkReply.NoError:
        return
    tempstr = str(tempreply.readAll())
    tempdata = json.loads(tempstr)
    if tempdata['temp'] == '':
        return
    if Config.metric:
        s = Config.LInsideTemp + \
            "%3.1f" % ((float(tempdata['temp']) - 32.0) * 5.0 / 9.0)
        if tempdata['temps']:
            if len(tempdata['temps']) > 1:
                s = ''
                for tk in tempdata['temps']:
                    if s != '':  # Add a newline only if s already has a value
                        s+= '\n'  # Add a newline only if s already has a value
                    s += ' ' + tk + ':' + \
                        "%3.1f" % (
                            (float(tempdata['temps'][tk]) - 32.0) * 5.0 / 9.0)
    else:
        s = Config.LInsideTemp + tempdata['temp']
        if tempdata['temps']:
            if len(tempdata['temps']) > 1:
                s = ''
                for tk in tempdata['temps']:
                    if s != '':  # Add a newline only if s already has a value
                        s+= '\n'  # Add a newline only if s already has a value
                    s += ' ' + tk + ':' + tempdata['temps'][tk]
    temp.setText(s)
```
I can't stress this enough, DOCUMENT YOUR CHANGES! Updating the clock will break this and you will have to reapply the changes manually.

Finally you'll need to **set up the gpio pins** that your additional sensors are on:
`sudo nano /boot/config.txt`

Look for:
`dtoverlay=w1-gpio,gpiopin=4` 

and add your pins like so:
`dtoverlay=w1-gpio,gpiopin=4,gpiopin=17`

**DISCLAIMER**
I did this off the top of my head and in the browser without testing. I am also on limited sleep so..... Let me know if you have issues and I'll hook up several sensors to mine and test them

xxxDrewedxxx, 2019-03-25:
Thanks TogaTown, that is exactly what I needed.  I didn't have to change the gpio pins.  
[image]

togatown, 2019-03-25:
Nice variation! I like your icon set. Are you using an actual sensor for your outside temp instead of the api?

xxxDrewedxxx, 2019-03-25:
Yes I am.  However it is currently sitting on my desk connected to the bread board (hense why the indoor and outdoor are almost the same).  It is going to stay that way for a day or two to make sure everything stays happy, then it’s cable will get soldered in and the sensor will get sent outside.

n0bel, 2019-07-21:
Closing as no more activity

closed 2019-07-21 by n0bel