n0bel/PiClock #154: Frame 3

garry51, 2019-04-05:
Can someone help please I have a pi weather station that has been working for about a year now it generates Rrd graphs as png images I would like to show them on frame 3 of my pi clock but alas I no nothing about style sheets so can someone show me how to put 4 png images on frame 3 please.
Thanks

togatown, 2019-04-06:
Create a label as a sub element of your frame 3:

```
imgcont = QtGui.QLabel(frame3)
imgcont.setObjectName("imgcont")
imgcont.setStyleSheet("imgcont { background-color: transparent; }")
imgcont.setGeometry(0, 0, 720 * xscale, 900 * yscale)
```
Then set the image to appear in the label:

```
myIconMap = QtGui.QPixmap( "pathandfilename.png")
imgcont.setPixmap(myIconMap.scaled(imgcont.size(), Qt.IgnoreAspectRatio, Qt.SmoothTransformation))
```

garry51, 2019-04-07:
Thanks for the help. I will try it today.

Cheers.

garry51, 2019-04-07:
Thank's it works a treat you have made an old guy very happy.
cherrs Garry

garry51, 2019-04-07:
The graph image displays ok but alas it wont update. I am sshpassing the image every 5mins to the clock (all working ok) but the clock only displays the image it has on boot up no refresh. Any ideas.

Thanks Garry

togatown, 2019-04-08:
You'll have to create a routine to update the image every so often or apply it inside one of the existing loops.

garry51, 2019-04-11:
Its working ok now thanks for the help.

closed 2019-04-11 by garry51