n0bel/PiClock #155: Hiding Google maps box (refer picture)

mdhadi, 2019-04-18:
Hi..

How to hide those grey box (Google Maps) on left side & the box on the right side ? Which line need to commented in order to get make it work ? The code must be in **PyQtPiClock.py** right ?I just want to show clock only since I don't use CC for Google Map API.

[image]

togatown, 2019-04-18:
If all you are interested in is the clock and the date, you are better off re-coding the project to remove all of the weather and map retrieval routines. What is CC?

mdhadi, 2019-04-18:
Hi..

CC = Credit Card

Seems Google charging USD300 is too much for currency conversion, currently on free trial for 365 days.

After lot of attempt, I manage to find the  line & commented out. So, my question is answered. Refer new image:

I just keep this thread open for time being.

[image]

togatown, 2019-04-18:
You should update to the new version and use Dark Sky and Mapbox. Both are free with limited use which exceeds to need of the clock.

mdhadi, 2019-04-18:
> You should update to the new version and use Dark Sky and Mapbox. Both are free with limited use which exceeds to need of the clock.

Okay

mdhadi, 2019-04-18:
I just register with Dark Sky and Mapbox, update the API as per guide. So far look goods. Only need to adjust the layout. Overlapping like issue #125 .

n0bel, 2019-07-21:
closing for no activity

closed 2019-07-21 by n0bel