n0bel/PiClock #156: Layout settings

MIUIArabia, 2019-04-23:
Hello, 

first of all, GREAT WORK. this looks very nice and very handy and I'm happy we have this project.

I'm sorry if someone posted this before, I just didn't find my answer.
I don't understand a thing in Python language which made it very hard for me to finish the work alone.

I'm seeking help in the following matters:

[image]

1- Get the icon and temp down a bit since they're sticking out of the screen border or maybe smaller?

2- Shrink the info under the temp to make it nicer and declutter the area.

3- Remove the reserved line for the internal sensor and get the clock in center or a bit down to stay away from the top text.

It would be a great help to show me where I need to edit or delete to have this done.

Thank you!

MIUIArabia, 2019-04-23:
So, I've solved 1 and 2 just by changing the clock size. it didn't make a lot of a difference so it's ok by me.

Issue #1 is still there. the icon is getting out of frame and the text size is not as desired.

togatown, 2019-04-24:
The icon in the top left is displayed in wxicon. You can change to position or size on the line that reads:
```
wxicon.setGeometry XPos, YPos, Width, Height
```

Here are the PyQt object type and names. 
```
Frame1:
├── QtGui.QFrame:
│	├── squares1 - Grid, weather maps
│	└── squares2 - Grid, forecast
│
├── QtCore.QRect
│	├── radar1rect - Radar 1
│	└── radar2rect - Radar 2
└── QtGui.QLabel:
	├── datex - Main Date Display
	├── wxicon - Current Conditions Icon in top left
	├── wxdesc - Current Conditions Description ie. Clear, Raining
	├── press - Current Conditions Pressure
	├── humidity - Current Conditions Humidity
	├── wind - Current Conditions Wind
	├── wind2 - Current Conditions Feels Like
	├── wdate - Last Updated
	├── bottom - Sun Rise/Set
	├── lab - Array of forecast
	├── forecast - daily array
	├── attribution - DarkSky.net attribution
	├── clockface - Main Time Display
	├── temp - Indoor Temp
	└── temper - Current Conditions Temperature

Frame2:
├── QtCore.QRect
│	├── radar3rect - Radar 3
│	└── radar4rect - Radar 4
└── QtGui.QLabel:
	├── datex2 - date display
	├── datey2
	├── wxicon2
	├── wxdesc2
	├── tepmer2
	└── attribution2 - DarkSky.net attribution
```

MIUIArabia, 2019-04-27:
Thank you very much for the help.

closed 2019-04-27 by MIUIArabia