n0bel/PiClock #157: Radar Images have alot of noise

matg333, 2019-04-24:
I am getting alot of noise on the radar images.  The noise looks like theres an old tv screen with static.  Im using the Dark sky and mapbox api's.  Is their anything I can do to clear up the radar images?  Thanks!

SerBrynden, 2019-05-29:
I see that sometimes, as well.  It's either a problem with the RainViewer.com API, from which the radar image are taken, or a problem with how Qt tiles the images together.  I think it's the first one and you'll have to take the issue to RainViewer.com.  Also, occasionally you'll see circular clumps of noise centered around the radar sites (you can lookup where they are).  This usually occurs on clear days when I think the Weather Service engineers crank up the radar sensitivity, resulting in false returns from low-altitude dust, birds, or moist air near the site.  That's my theory.

xenon462, 2019-05-29:
This is what they write on https://www.rainviewer.com/help.html

n0bel, 2019-07-21:
Closing as no further activity

closed 2019-07-21 by n0bel

SerBrynden, 2023-04-16:
See https://github.com/n0bel/PiClock/issues/241#issuecomment-1510003323