n0bel/PiClock #160: Mapbox Maps Not There

maxheadspace, 2019-06-08:
I seem to be the only person using MapBox, as I can find very few references to it in any of the discussions.  Anyway, I obtained my key and have it inserted in the ApiKeys.py file and have no errors.  But when I run PiClock, everything comes up except the MapBox maps.  I'm not sure if there is a setting that must tell PiClock to use MapBox, but I find nothing on that topic in the instructions.  I did find the one line about MapBox which was set to =0, and I set that to =1, but that did not solve the map issue.  I put a hashtag in front of the googlemaps API line, but that didn't help either.  Surprisingly the radar is actually showing clouds moving past, but there's no map underneath, just the gray box.  One possible wrinkle is that I am currently in South Korea.  Maybe MapBox doesn't work in South Korea?  Thanks for help!

BertLindeman, 2019-06-08:
Hi (How should I address you?)

Did you base your config upon
`/home/pi/PiClock/Clock/Config-Example.py` ?

At about line 48 there is:
```
usemapbox = 0 # Use Mapbox.com for maps, needs api key (mbapi in ApiKeys.py)
```

Set it to `1` in the config that you use, to use mapbox.

Bert

maxheadspace, 2019-06-09:
Thank you for the reply, Bert.  Yes, I changed that setting from 0 to 1, but still no maps.  It's odd, because the radar images of rain(?) are crossing the map boxes, so the radar part works, but there are no maps underneath the radar images.  

I tried obtaining a GoogleMaps API, but it will not accept my credit card because it is was issued in USA and I am living now in Korea, plus it required me to sign up for cloud service, which I do not want.

Perhaps MapBox does not work outside the USA?

Thanks again, Bert.

Max

BertLindeman, 2019-06-09:
Hi Max ;-) 

Did you look into the log file? 
Search for **mapbox** and you'll find the URL (webaddress) that the clock uses to get the map pictures.

Looks like 
```
https://api.mapbox.com/styles/v1/mapbox/satellite-streets-v10/static/4.627072,51.892625,1,0,0/166x146?access_token=
```
I removed my api token of course....

You can copy that URL from your `PyQtPiClock.*.log` and paste that into your webbrowser and see if that produces a picture or an error (maybe about the coordinates? Did you specify latitude/longitude in the correct order?).

I am in Europe and mapbox is OK here.
This URL shows a small map of a part of Asia (use your own access token):
```
https://api.mapbox.com/styles/v1/mapbox/satellite-streets-v10/static/126.93593538115356,37.50972584293751,1,0,0/166x146?access_token=
```
[image]

maxheadspace, 2019-06-15:
Thanks Bert.  That sounds like it will identify the problem, but what commands do I use to review the log file?  Also, is it possible that I didn't configure the API token correctly?  It requires the user to define the "scopes" before creating the token, but I have no idea what the scopes represent and there isn't a ready definition available on the MapBox page.

Appreciate your help.  Thanks Bert!

Max

BertLindeman, 2019-06-15:
Hi Max, 

Would be good to know what operating system you use and what experience you have on that platform.
Now I have to guess ;-)
Assuming you use some kind of Linux for now (raspbian?) and no relevant changes on `/home/pi/PiClock/startup.sh` 

1. **log directory and filename**

The logfile is written by  `startup.sh` on line 131 like this:
```
python -u PyQtPiClock.py >PyQtPiClock.1.log 2>&1
```

So the latest logfile would be written into directory `/pi/PiClock//home` as
`PyQtPiClock.1.log`

2. **Search the log file**
Use a terminal to access your PyClock system and use the commands ```less``` ```more``` and/or ```grep```
to show the log.
As an example here use grep to only show log-lines that contain the string **`mapbox`**:
```
cd  /pi/PiClock//home
grep -i mapbox PyQtPiClock.1.log
```

End of lecture ;-)

Hope it's enough to get a step further.
Bert

maxheadspace, 2019-06-23:
Thanks for the suggestion, Bert.  Your recommended grep line came back with "no such file."  I'm using Raspbian Jessie, which may be different from what you suggested.  

The instructions are at this link:  https://hackaday.io/project/6184/instructions

When I boot up and open the terminal screen I am already at  pi@PiClock:/ $

And if I try any cd, such as /pi/PiClock//home, it comes back with "no such directory."  Hopefully this makes sense to you.

Thanks again, Bert!

Max

maxheadspace, 2019-06-23:
Correction, on startup it boots to:   pi@PiClock:~ $

But the grep command still returns "no such file".  When I change to folder PiClock, I get the same response, and when I change to PiClock/Clock I still get a "no such file."

Not sure if that adds clarity.

Thanks again!

maxheadspace, 2019-06-23:
Screen shot of my last attempt:
[image]

BertLindeman, 2019-06-23:
Oke, Max.

The `cd` command got wrangled up (my fault)
Somehow the `/home` part got shifted to the end while it should be in front...

Keep in mind that directory names and filenames are **case sensitive**
so `cd piclock` is different from `cd PiClock`

Trying to propose better commands ;-)
I added an `ls` command to see which files are there:

```
cd  /home/pi/PiClock/Clock/
ls -l 
grep -i mapbox PyQtPiClock.1.log
```

I assume you will get more than one line of hits on mapbox.
The lines may be wrapped as they are usually longer than the terminal width.
Copy the whole line starting with `https` so skip this part at the beginning of the line:
`map base url: `

Hope I got it correct this time ;-)

Bert

PS

If you still get 'no file or directory' you might try this command:

```
find /home/pi -name "PyQtPiClock*.log" -ls
```

n0bel, 2019-07-21:
closing as no further activity.

closed 2019-07-21 by n0bel