n0bel/PiClock #161: Google api suddenly not working

GrampDamnpus, 2019-06-25:
Starting today my clock wont pull maps. I have it set up with google maps api, I checked the dashboard and as far as I can tell everything is good to go. 

When I check the log I dont get any errors, but if I put the url for the map into a browser I get:
The Google Maps Platform server rejected your request. This API key is not authorized to use this service or API.

I checked 4 times and the staticmap api is enabled and seems good, I have 300 in credit and no outstanding bill. I have switched to mapbox for the day, and will try google again tomorrow maybe its something on their end, unless someone knows something that changed?

togatown, 2019-06-27:
If you are taking the URL from the logs, pasting into a browser and getting an error, the issue isn't with PiClock but with the Google API. I would get the URL working in the browser first.

If Mapbox works for you, why bother with the Google API at all?

GrampDamnpus, 2019-06-27:
> If you are taking the URL from the logs, pasting into a browser and getting an error, the issue isn't with PiClock but with the Google API. I would get the URL working in the browser first.
> 
> If Mapbox works for you, why bother with the Google API at all?

100% is a problem with google, never said different, I was more checking to see if I was the only one with this issue or something had changed with the api functions and the clock would need an update. 
Sadly I wouldnt know where to begin trouble shooting the api. 

As for why I would use google when mapbox works, the quick answer is because. The long answer is because I already had a google api from legacy and the maps are cleaner than the mapbox ones.  Im gonna go test some stuff that I can and see what I come up with.

togatown, 2019-06-28:
I would test it for you but after I assured myself Mapbox was working I blew away my Google account and never looked back.

I agree the Google satellite map is cleaner looking but I adapted rapidly. Have you looked at the styling to see if anything strikes you?

n0bel, 2019-07-21:
closing as no further activity

closed 2019-07-21 by n0bel