n0bel/PiClock #165: Feature request: Tidal information

vanosg, 2019-07-29:
@n0bel this is an amazing project, you did a fantastic job developing it, and an even better job documenting it! Hats off for that.

Do you think there is any chance of integrating tidal information onto a screen? ie, high tide/low tide? If you could rough in a place on the GUI, I'd be happy to code up something to parse an API. NOAA offers an API that might be of use via https://tidesandcurrents.noaa.gov/api/ as an example. Please let me know if I can assist! Thanks again

togatown, 2019-07-30:
@vanosg what kind of data are you looking for? [Just High/Low times?](https://tidesandcurrents.noaa.gov/api/datagetter?units=english&format=json&station=8721604&date=today&time_zone=lst_ldt&interval=hilo&product=predictions&datum=MLLW)

vanosg, 2019-07-30:
Yes, just the next high and low (or maybe two sets of high and low, if there is space for it, but I think that may be ambitious)

skipwill, 2023-02-20:
I added another hand to the PiClock to track the tide movement. 12 o’clock is high tide and 6 o’clock is low tide. The hand moves as it tracks the tide.

vanosg, 2023-02-22:
This sounds amazing! Is the code available somewhere?

skipwill, 2023-04-11:
Let me get the openweather conversion done first and I’ll post the tide clock hand code. I am having to convert to qt5 and python 3 as qt4 seems to not be available anymore.