n0bel/PiClock #167: Reading from Database for Temp/Pressure/wind.......

billlariv, 2019-07-30:
I am attempting to modify my PiClock to read from a mysql db to get the information that is displayed in the top left corner of PiClock.  I have changed to a digital clock and moved the radar squares below the digital clock.  So that gives me plenty of room on the left edge of the screen for my data.  I have a python script that retrieves the data I am wanting to display.  (see below).  I have attempted to create function with it, and inserted in PyQtPiClock.py.  Creating global variables for the datapoints and have tried to use those variable names.  No luck.  Any help would be greatly appreciated.  I am technical, but not a programmer.

#!/usr/bin/python
import sys
import re
import mysql.connector

from mysql.connector import Error
try:
   mySQLconnection = mysql.connector.connect(host='localhost',
                             database='weewx-db',
                             user='weewx-user',
                             password='weewx-password')

   sql_select_Query = "select outTemp from archive order by dateTime Desc limit 1;"
   cursor = mySQLconnection .cursor()
   cursor.execute(sql_select_Query)
   WEEWX_TEMP = cursor.fetchone()
   print ("Temp")
   print "%.1f" % WEEWX_TEMP +" F"
   print("")

except Error as e :
    print ("Error while connecting to MySQL", e)

try:
   sql_select_Query = "select windSpeed from archive order by dateTime Desc limit 1;"
   cursor = mySQLconnection .cursor()
   cursor.execute(sql_select_Query)
   WEEWX_WINDSPEED = cursor.fetchone()
   print ("Wind Speed")
   print"%.0f" % WEEWX_WINDSPEED +" mph"
   print ("")

except Error as e :
    print ("Error while connecting to MySQL", e)

try:
   sql_select_Query = "select windDir from archive order by dateTime Desc limit 1;"
   cursor = mySQLconnection .cursor()
   cursor.execute(sql_select_Query)
   WEEWX_WINDDIR = cursor.fetchone()
   DIRECTION = "N"

   if 0 <= WEEWX_WINDDIR <= 22.5:
    DIRECTION = "N"

   if 22.6 <= WEEWX_WINDDIR <= 67.5:
    DIRECTION = "NE"

   if 67.6 <= WEEWX_WINDDIR <= 112.5:
    DIRECTION = "E"

   if 112.6 <= WEEWX_WINDDIR <= 157.5:
    DIRECTION = "SE"

   if 157.6 <= WEEWX_WINDDIR <= 202.5:
    DIRECTION = "S"

   if 202.6 <= WEEWX_WINDDIR <= 247.5:
    DIRECTION = "SW"

   if 247.6 <= WEEWX_WINDDIR <= 292.5:
    DIRECTION = "W"

   if 292.6 <= WEEWX_WINDDIR <= 337.5:
    DIRECTION = "NW"

   if 337.6 <= WEEWX_WINDDIR <=360.0:
    DIRECTION = "N"

   print ("Wind Dir")
   print(DIRECTION)
   print ("")

except Error as e :
    print ("Error while connecting to MySQL", e)

try:
   sql_select_Query = "select outHumidity from archive order by dateTime Desc limit 1;"
   cursor = mySQLconnection .cursor()
   cursor.execute(sql_select_Query)
   WEEWX_HUMIDITY = cursor.fetchone()
   print ("Humidity")
   print"%.0f" % WEEWX_HUMIDITY +" %"
   print ("")

except Error as e :
   print ("Error while connecting to MySQL", e)

try:
   sql_select_Query = "select rain from archive order by dateTime Desc limit 1;"
   cursor = mySQLconnection .cursor()
   cursor.execute(sql_select_Query)
   WEEWX_RAIN = cursor.fetchone()
   print("Rain Today")
   print"%.2f" % WEEWX_RAIN +" in"
   print("")

except Error as e :
    print ("Error while connecting to MySQL", e)

try:
   sql_select_Query = "select rainRate from archive order by dateTime Desc limit 1;"
   cursor = mySQLconnection .cursor()
   cursor.execute(sql_select_Query)
   WEEWX_RAINRATE = cursor.fetchone()
   print("Rain Rate")
   print"%.2f" % WEEWX_RAINRATE +" in/hr"
   print("")

except Error as e :
    print ("Error while connecting to MySQL", e)

try:
   sql_select_Query = "select heatindex from archive order by dateTime Desc limit 1;"
   cursor = mySQLconnection .cursor()
   cursor.execute(sql_select_Query)
   WEEWX_HEATINDEX = cursor.fetchone()
#   print("Heat Index")
#   print"%.1f" % WEEWX_HEATINDEX
#   print("")

except Error as e :
   print ("Error while connecting to MySQL", e)

try:
   sql_select_Query = "select UV from archive order by dateTime Desc limit 1;"
   cursor = mySQLconnection .cursor()
   cursor.execute(sql_select_Query)
   WEEWX_UVINDEX = cursor.fetchone()
   print ("UV Index")
   print"%.1f" % WEEWX_UVINDEX 
   print ("")

except Error as e :
    print ("Error while connecting to MySQL", e)

try:
   sql_select_Query = "select windchill from archive order by dateTime Desc limit 1;"
   cursor = mySQLconnection .cursor()
   cursor.execute(sql_select_Query)
   WEEWX_WINDCHILL = cursor.fetchone()
#   print("Wind Chill")
#   print"%.1f" % WEEWX_WINDCHILL
#   print("")

except Error as e :
    print ("Error while connecting to MySQL", e)

finally:
    #closing database connection.
    if(mySQLconnection .is_connected()):
        mySQLconnection.close()

if WEEWX_TEMP > 50:
        print("Heat Index")
	print("%.1f" % WEEWX_HEATINDEX +" F")
        print("")
else:
        print("Wind Chill")
	print("%.1f" % WEEWX_WINDCHILL +" F")
        print("")

billlariv, 2019-09-06:
Is there a single place in PyQtPiClock.py that I can replace temp, pressure.... with my variable?

togatown, 2019-09-09:
Looks like you are trying to use data from your weather station perhaps? There is no 'single place' because you are looking to replace quite a bit of info. Search the code for "def wxfinished()", this is the function that fills in the data you are concerned with. Then inside of def wxfinished(), search for "setText". This is how the text is set in each of the labels.

I personally would allow the routine to run and then call your function at the end of it to retrieve and fill your own values.

billlariv, 2019-09-26:
Thanks for the response.  Here is what I tried.
def wxfinished()
   getlocalwx()
   temper=WEEWX_TEMP

Thinking I can set the variable of the outside temperature in PiClock to equal that which was read from the weewx database.  It did not work.

togatown, 2019-09-27:
getlocalwx should be the last thing in wxfinished. In it you would set your label to your desired value. 

```
temper.setText(yourvalue)
```

billlariv, 2019-10-19:
togtown thanks for the reply.  I have changed directions sort of.  Instead of reading from DB I am tailing a file that has live data.  Weewx takes a while to archive data to db.  So I am using cumulus realtime.txt that gets new data about every second.  I have my variables getting the values from the file.  I am now looking at how to update the information from temp down on the left side of the screen when the variables change.  I am running my function with threading.
[image]

billlariv, 2019-10-19:
def getlocalwx():
    while True:
        filename = '/var/www/weewx/realtime.txt'
        f = subprocess.Popen(['tail','-F',filename],\
        stdout=subprocess.PIPE,stderr=subprocess.PIPE)
        p = select.poll()
        p.register(f.stdout)
        if p.poll(1):

            wordList = f.stdout.readline().split(' ')
            WEEWX_TEMP = float(wordList[2])
            WEEWX_HUMIDITY = float(wordList[3])
            WEEWX_HEATINDEX = float(wordList[41])
            WEEWX_WINDSPEED = float(wordList[6])
            WEEWX_WINDGUST = float(wordList[40])
            WEEWX_WINDDIR = wordList[11]
            WEEWX_RAINRATE = float(wordList[8])
            WEEWX_RAIN = float(wordList[9])
            WEEWX_WINDCHILL = float(wordList[24])

            if float(WEEWX_TEMP) > 50:
                WEEWX_LFEELS = "Heat Index"
                WEEWX_FEELSLIKE = WEEWX_HEATINDEX
            else:
                WEEWX_LFEELS = "Wind Chill"
                WEEWX_FEELSLIKE = WEEWX_WINDCHILL

            WEEWX_TEMP = str(WEEWX_TEMP)

            temper.setText(str(WEEWX_TEMP) +  u'°F')
            press.setText(Config.LPressure + "\n" + '%.2f' % pressi(f['pressure']) + 'in')
            humidity.setText('Humidity' + ' ' + str(WEEWX_HUMIDITY) + ' %')
            wd = WEEWX_WINDDIR
            if Config.wind_degrees:
                wd = str(f['windBearing']) + u'°'
            wind.setText("Wind " + wd + ' ' + '%.0f' % WEEWX_WINDSPEED + ' mph' + "\n" + "Gust" + ' ' + '%.0f' % WEEWX_WINDGUST + ' mph')
            wind2.setText(WEEWX_LFEELS + "\n" + str(WEEWX_FEELSLIKE) +  u'°F')
            wdate.setText('Rain - ' + str(WEEWX_RAIN) + "\n" + 'Rate - ' + str(WEEWX_RAINRATE))
        wxfinished()
        time.sleep(2)

getlocalwxfirst()
WEEWX_LFEELS = "Heat Index"
WEEWX_FEELSLIKE = str(WEEWX_HEATINDEX)
thread = GETLOCALWXD(getlocalwx)
thread.start()

billlariv, 2019-10-23:
I think I have it figured out.  Using weewx opensource weather station software and cumulus extension we are live.  Weewx software is getting data from the weather station and the cumulus extension is getting the data as it if fed to the pi.  Cumulus is then updating a file named realtime.txt with the latest data from the weather station.  It is a one line file so easy enough to read.  I have attached my hack at python to the issue.

[PyQtPiClock.txt](https://github.com/n0bel/PiClock/files/3759528/PyQtPiClock.txt)

closed 2019-10-23 by billlariv