n0bel/PiClock #170: No more radar images?

ScottChapman, 2019-09-02:
I seem to no longer be getting any radar images. For the past several weeks I've notices that my maps always show no radar images, even when it is actually raining outside (and clearly I can see rain radar images using a weather app).

Any ideas on trouble shooting this? I have restarted the Pi several times, and I do see the time updating under the radar images...

feh123, 2019-09-02:
No ideas but just to comment that Rainviewer.com is delivering accurate radar images for the UK. Nothing noticeably wrong for the past month (even the black areas are very rare now). I have a Pi 3B - PiClock is its sole job, no sensors, LEDs.... the most basic setup.

ScottChapman, 2019-09-02:
Pulled latest (usually don't unless there is something specific I am looking for). Rebooted, s'all good man.

closed 2019-09-02 by ScottChapman