n0bel/PiClock #179: [Question] DarkSky API now that Apple has stepped in

gavO852, 2020-04-01:
How is this project going to adapt to this change? The DarkSky API is going to be shutdown at the end of 2021 because Apple just bought them today. Is it already possible to change the API used to another service? Just wondering! 

Have a great day!

_Link to DarkSky post : https://blog.darksky.net_

spfunnell, 2020-04-02:
I was wondering the same thing.  What are the alternatives?  Thankfully there's a bit of time to hopefully get it sorted out.

n0bel, 2020-04-02:
Worry not.   We've been through this before (Weather Underground) we'll do it again.
Any suggestions on free weather api's are appreciated.
Requirements:
1) Free
2) Reasonable Rate limits (allowing for 30 min or 10 minute updates)
3) Frequent Data Updates (15-30 minute updates) (Some APIs don't update their data but once every 3 -4 hours.  This is a pain when showing current conditions.
4) Current Conditions
5) 10 Day Forecast (7 day minimum)
6) Hourly Forecast

vanhalen26505, 2020-04-02:
Kevin, thanks so much for continuing to support this fantastic app for all
of us. Much appreciated!!!

BertLindeman, 2020-04-02:
Hi all,

Maybe this page is a starting point:

https://rapidapi.com/blog/access-global-weather-data-with-these-weather-apis/

Came there by searching for: free weather forecast data api

Obviously DarkSky info in there will be changed end 2021.

(Thank you, Kevin)

Bert

ScottChapman, 2020-04-02:
Thanks for keeping up on this. Is it going to be possible to continuing using DarkSky (or weather underground for that matter) is one has an account and API keys?

maserowik, 2020-04-02:
https://rapidapi.com/blog/access-global-weather-data-with-these-weather-apis/
any of thee work?
Mike- N3BSQ

   - Free
   - Current Conditions

gavO852, 2020-04-02:
> Worry not.   We've been through this before (Weather Underground) we'll do it again.
>
> Any suggestions on free weather api's are appreciated.
>
> Requirements:
>
> 1) Free
>
> 2) Reasonable Rate limits [...]

Maybe openweathermap.org?

Knackster, 2020-04-03:
Can any data be scraped from the NWS?
https://www.weather.gov/jetstream/ridge_download

Chris

merky1, 2020-04-03:
Here are the ones I know...

Aeris - free, but 90 day key is less than optimal
Nws - free, but us centric and API requires a lot of testing.  Bad docs, limited public info
Open weather - free, but updates are 30-45 mins stale

Maybe weather 2020 or something similar.

n0bel, 2020-04-03:
https://www.weatherbit.io/pricing

To me OpenWeatherMap being not updated is a non starter (Pricing page says free is updated < 2 hours, meaning up to 2 hours old information for current weather)

NWS, nope, not global, but only for USA.
Aeris, yeah, no not free.

wttr.in seems promising....
https://github.com/chubin/wttr.in
https://wttr.in/?format=j1
My testing of that so far has resulted in 1 hour old data.

wttr gets its data from https://www.worldweatheronline.com/developer/api/pricing2.aspx
$5.00/month.. well maybe     I still need to test.

Please feel free to test various apis and see what their various stale-ness is.

merky1, 2020-04-03:
Weatherbit has the current conditions in the api docs.  Need to set up a nightly script for an evaluation.  I’ll probably do a test over the weekend to compare aeris, dark sky, and weatherbit.

For those who have current keys, I’d stay away from changing providers until next year.  Dark sky has been ideal, and hopefully another team will step up and provide a similar environment.  Sadly, weather data seems to be moving in the mobile phone direction, and open API’s will be harder to find.  It seems that free impacts the availability of current conditions, and that has been a huge issue for “us.”

merky1, 2020-04-07:
sorry, had a hardware failure that took way too long to diagnose / resolve.

So looking at the weatherbit.io free offering, there is a similar 1 hour lag to data updates.

Data update delay: 1 hour

Knackster, 2020-04-07:
A delay of one hour for weather data is a non-starter for me.I used my
PiClock (thank you again Kevin) daily as an up-to-date weather tool vs
having to visit a web site. Our weather here in Texas can take a turn for
the worse in minutes.

Chris

merky1, 2020-04-07:
https://www.visualcrossing.com/resources/news/how-can-i-replace-the-darksky-api/  

Looks promising, if you can figure out how to get a key.

n0bel, 2020-04-08:
Did a bit of digging.
Their free tier allows for 50 results per day.   I believe that is like 50 calls per day, but I'm not sure.

https://www.visualcrossing.com/weather-data-editions

So this would mean at best about 30 minutes between calls.

xenon462, 2020-04-08:
https://rp5.ru/docs/xml/en
The weather forecasts cost $1

merky1, 2020-04-08:
> So this would mean at best about 30 minutes between calls.

It’s not clear, but the 50 restriction specifically calls out an excel macro, and that makes no sense to me.  I sent out some feeler emails / registrations, so we’ll see.  Worst case, open weather is done and can be used as a fallback.

Additionally, yahoo weather is an option, but again updates are 1hr.

merky1, 2020-04-08:
Hi Tom,

A few details to get you started:

Your API key:<key>

**Calls limit: 1000/day**

Selected plan: **Free**

n0bel, 2020-04-09:
sounds like some promising options  $1 per month isnt too bad either.

Lets see what the lag time is on these.

Also, does anyone know if METAR data is available for al airports in the world?   In the US it is of course used extensively.   It would work for current conditions.  Forecast data could be used from one of the laggy services.   I see my closest local airport has a 15 minute lag.

feh123, 2020-04-09:
Hi
Metars.com my PC shows my local and internationl airports in the UK. I do not see the weather data on my PC browser but UK data can be seen on my iOS metar app. My nearest Metar station is Stansted airport 23 ml away and I get a good set of data. Not sure how to get the lag time.

xenon462, 2020-04-09:
The latest observations in METAR format appear on the website once or twice per hour, approximately in 10 minutes after weather observations have been made at the weather station.
https://rp5.ru/docs/sgate/ru

xenon462, 2020-04-09:
Forecasts are completely updated on the website twice a day: at 05:30 and 17:30 UTC. The latest observational data in SYNOP format is available on the website eight times a day, every three hours: as a rule, by 0:30, 3:30, 6:30 9:30, 12:30, 15:30, 18:30 and 21:30 UTC. The latest observations in METAR format appear on the website once or twice per hour, approximately in 10 minutes after weather observations have been made at the weather station.
https://rp5.ru/docs/about/en

merky1, 2020-04-23:
2020-04-23 01:33:17 PM -0400 info: openweather : 57.51 : Thu Apr 23 2020 13:31:16 GMT-0400 (EDT)
2020-04-23 01:33:17 PM -0400 info: darksky : 63.04 : Invalid Date
2020-04-23 01:33:17 PM -0400 info: climacell : 58.44 : Thu Apr 23 2020 13:33:17 GMT-0400 (EDT)
2020-04-23 01:33:23 PM -0400 info: usg : 57.2 : Thu Apr 23 2020 13:15:00 GMT-0400 (EDT)

still cleaning up the times as you can see, but from what I can see, climacell seems to be in line with USG and openweather.  I will work on the observation times to validate.

merky1, 2020-05-02:
Still testing, but climacell seems stable and accurate enough for use.  I added the logic to my clock, and it’s been running for 24 hours.  I’m still mapping the weather codes to icons, but the current conditions came across fine.  I’d pull my changes in, but I forked off into nodeJS land...

http://climacell.co/pricing

n0bel, 2020-05-03:
Thanks for all your work, testing, and helping to pick our next api.

I'll start on the piclock api changes for climacell

lmkwxman, 2020-05-10:
For the US, the National Weather Service does have an API.  The link is https://api.weather.gov/

Here is a link to the documentation for the NWS API.  https://weather-gov.github.io/api/general-faqs.

-Mark

merky1, 2020-05-21:
2 major issues with noaa / nws data source.

1 it is US specific.  
2 the current observations are finicky, and have a low update rate.  I’ve pulled an overnight run against weather.gov, and the results are meh at best :

2020-05-04 07:42:44 AM -0400 info: openweather : 58.53 : 0 : Clear
2020-05-04 07:42:44 AM -0400 info: climacell : 60.35 : 0 : mostly_clear
2020-05-04 07:42:50 AM -0400 info: usg : 59 : 28
2020-05-04 07:52:44 AM -0400 info: darksky : 59.82 : 0 : Clear
2020-05-04 07:52:44 AM -0400 info: openweather : 59.05 : 0 : Clear
2020-05-04 07:52:44 AM -0400 info: climacell : 60.35 : 0 : mostly_clear
2020-05-04 07:52:47 AM -0400 info: usg : 60.8 : 18
2020-05-04 08:02:44 AM -0400 info: openweather : 59.05 : 10 : Clear
2020-05-04 08:02:44 AM -0400 info: darksky : 60.41 : 0 : Clear
2020-05-04 08:02:44 AM -0400 info: climacell : 61.93 : 0 : clear
2020-05-04 08:02:46 AM -0400 info: usg : 60.8 : 28
2020-05-04 08:12:44 AM -0400 info: darksky : 61.21 : 0 : Clear
2020-05-04 08:12:44 AM -0400 info: openweather : 59 : 0 : Clear
2020-05-04 08:12:44 AM -0400 info: climacell : 61.93 : 0 : clear
2020-05-04 08:12:46 AM -0400 info: usg : 60.8 : 38
2020-05-04 08:22:44 AM -0400 info: openweather : 59.45 : 0 : Clear
2020-05-04 08:22:44 AM -0400 info: darksky : 61.57 : 0 : Clear
2020-05-04 08:22:44 AM -0400 info: climacell : 61.93 : 0 : clear
2020-05-04 08:22:47 AM -0400 info: usg : 32 : 8
2020-05-04 08:32:44 AM -0400 info: darksky : 61.94 : 0 : Clear
2020-05-04 08:32:44 AM -0400 info: openweather : 60.01 : 0 : Clear
2020-05-04 08:32:44 AM -0400 info: climacell : 62.83 : 0 : mostly_clear
2020-05-04 08:32:46 AM -0400 info: usg : 62.6 : 18
2020-05-04 08:42:44 AM -0400 info: darksky : 62.62 : 0 : Clear
2020-05-04 08:42:44 AM -0400 info: openweather : 60.31 : 0 : Clear
2020-05-04 08:42:44 AM -0400 info: climacell : 62.83 : 0 : mostly_clear
2020-05-04 08:43:06 AM -0400 info: usg : 32 : 8
2020-05-04 08:52:44 AM -0400 info: openweather : 60.98 : 2 : Clear
2020-05-04 08:52:44 AM -0400 info: darksky : 63.26 : 0 : Clear
2020-05-04 08:52:44 AM -0400 info: climacell : 62.83 : 0 : mostly_clear
2020-05-04 08:52:54 AM -0400 info: usg : 62.6 : 18
apitest-2020-05-04.log lines 186-216/216 (END)

32 is a null value pull that I didn’t wrap (lack of care). The 3rd column is minutes behind now.  

The NOAA/NWS api is awesome for forecasts, and I use it for that.  The current conditions has always been laggy and intermittently returns nulls for values.  It is less reliable than open weather for this reason.

gavO852, 2020-06-02:
As it looks like this thread has reached its answer, I'll be closing this issue, with a new API being selected. Thanks n0bel for all that you do!

closed 2020-06-02 by gavO852

ScottChapman, 2020-06-02:
I think we need a new issue to track the climacell work then?

gavO852, 2020-06-02:
I'll leave that up to n0bel!

n0bel, 2020-06-03:
moved to https://github.com/n0bel/PiClock/issues/185