n0bel/PiClock #185: Darksky API Replacement

n0bel, 2020-06-03:
After a bunch of research carried out by the great people helping with this project, a new provider has been selected, as pointed out in this issue comment:

https://github.com/n0bel/PiClock/issues/179#issuecomment-622951747

We'll have a new version soon.

lhloftin47, 2020-06-09:
Some of us are impatient. Any idea how soon, soon is?  :-)

merky1, 2020-06-20:
The climacell API is different, in that you have to add fields to the query.  You also need to pass the cc api key as a http header.  The JSON return is pretty easy to parse.

[Climacell.co](https://developer.climacell.co/v3/reference)

```
async function currentCcObs(){
	var url = 'https://api.climacell.co/v3/weather/realtime?lat=' + settings.lat + '&lon=' + settings.lon + '&unit_system=us&fields=temp%2Cfeels_like%2Chumidity%2Cwind_speed%2Cmoon_phase%2Cweather_code%2Csunrise%2Csunset%2Cwind_direction%2Cbaro_pressure'
	logger.info(url);
		
	var { body } = await getPromise({
		url: url,
		json: true,
		headers: {'User-Agent': 'piclockjs',
			'apikey' : settings.ccAppId,
			'accept' : 'application/json'
		}
	});
	parseCC(body);
}
```

jeder73, 2020-07-18:
Did you take a look at https://openweathermap.org/?

merky1, 2020-07-18:
Open weather is sadly unreliable, with data being up to 2 hrs stale, and sequential polls have regressive time stamps.  I kept it in my code, but found that I had to protect from differing time stamps too much.  Essentially, the lowest you can poll OW reliably is about once every 10 mins.  Even without the timestamp issues, the data is laggy at best.  Third column is behind now():

2020-06-18 08:12:38 PM -0400 info: openweather : 74.44 : 5 : Clouds
2020-06-18 08:12:38 PM -0400 info: darksky : 73.64 : 0 : Partly Cloudy
2020-06-18 08:12:38 PM -0400 info: climacell : 74.19 : 0 : cloudy

Crowlord, 2020-07-19:
I did try to jump onto your fork but im a total amatuer and its melting my brain. 

I guess I am waiting for this one to update :(

adamoutler, 2020-08-02:
What about weather.gov?  

`root@HLAB-A17:/mnt/c/Users/adamo# curl -H "User-Agent: aadfasdfasdfasdf@adamoutler.com" https://api.weather.gov/gridpoints/TOP/31,80/forecast
{
    "@context": [
        "https://geojson.org/geojson-ld/geojson-context.jsonld",
        {
            "@version": "1.1",
            "wx": "https://api.weather.gov/ontology#",
            "geo": "http://www.opengis.net/ont/geosparql#",
            "unit": "http://codes.wmo.int/common/unit/",
            "@vocab": "https://api.weather.gov/ontology#"
        }
    ],
    "type": "Feature",
    "geometry": {
        "type": "Polygon",
        "coordinates": [
            [
                [
                    -97.1089731,
                    39.766826299999998
                ],
                [
                    -97.108526900000001,
                    39.744778799999999
                ],
                [
                    -97.079846700000004,
                    39.745119500000001
                ],
                [
                    -97.08028680000001,
                    39.767167000000001
                ],
                [
                    -97.1089731,
                    39.766826299999998
                ]
            ]
        ]
    },
    "properties": {
        "updated": "2020-08-02T15:26:01+00:00",
        "units": "us",
        "forecastGenerator": "BaselineForecastGenerator",
        "generatedAt": "2020-08-02T15:58:22+00:00",
        "updateTime": "2020-08-02T15:26:01+00:00",
        "validTimes": "2020-08-02T09:00:00+00:00/P7DT16H",
        "elevation": {
            "value": 441.95999999999998,
            "unitCode": "unit:m"
        },
        "periods": [
            {
                "number": 1,
                "name": "Today",
                "startTime": "2020-08-02T10:00:00-05:00",
                "endTime": "2020-08-02T18:00:00-05:00",
                "isDaytime": true,
                "temperature": 78,
                "temperatureUnit": "F",
                "temperatureTrend": null,
                "windSpeed": "10 to 15 mph",
                "windDirection": "N",
                "icon": "https://api.weather.gov/icons/land/day/few?size=medium",
                "shortForecast": "Sunny",
                "detailedForecast": "Sunny, with a high near 78. North wind 10 to 15 mph, with gusts as high as 20 mph."
            },
            {
[... 172 lines cut ...]
                "temperatureTrend": null,
                "windSpeed": "5 to 10 mph",
                "windDirection": "S",
                "icon": "https://api.weather.gov/icons/land/day/tsra_hi,40/tsra_hi,20?size=medium",
                "shortForecast": "Chance Showers And Thunderstorms",
                "detailedForecast": "A chance of showers and thunderstorms. Mostly sunny, with a high near 87. Chance of precipitation is 40%."
            },
            {
                "number": 14,
                "name": "Saturday Night",
                "startTime": "2020-08-08T18:00:00-05:00",
                "endTime": "2020-08-09T06:00:00-05:00",
                "isDaytime": false,
                "temperature": 70,
                "temperatureUnit": "F",
                "temperatureTrend": null,
                "windSpeed": "5 mph",
                "windDirection": "S",
                "icon": "https://api.weather.gov/icons/land/night/tsra_hi,30?size=medium",
                "shortForecast": "Chance Showers And Thunderstorms",
                "detailedForecast": "A chance of showers and thunderstorms. Mostly clear, with a low around 70. Chance of precipitation is 30%."
            }
        ]
    }
}`

adamoutler, 2020-08-02:
OpenWeather API switching guide  https://openweathermap.org/darksky-openweather#match

scwestfisher, 2020-08-05:
I did do a quick and dirty fix to switch to openweathermap.org in my fork.
Steve

adamoutler, 2020-08-05:
I switched to https://github.com/merky1/PiClock because of far more activity.

merky1, 2020-08-06:
Keep in mind that my fork forgoes the python aspect, and is managed by a madman...

indymx, 2020-08-30:
Has this project died?

merky1, 2020-08-30:
If you need an immediate response, my fork is fully operational for the United States.  It uses nodejs, so it’s a complete rebuild of the pi, and my forks manager has a bad habit of testing updates in the main branch.  Also, my documentation is poor, so it’s really not ready for human use.  I have not gotten around to doc‘ing the settings file.  That and I’m tied to the US NOAA for forecasting and radar which makes things very US specific.

If your brave: https://github.com/merky1/PiClock

If you have the PiClock up and running today, I’d say stay until the darksky api turns off.  Hopefully there’s enough smart people here to fix that issue.

adamoutler, 2020-08-30:
I vouch for @merky1 repo.  I switched and am using it just fine.  The settings needs documentation.  But everything works.  Feel free to contribute, I did a bit.

N4IK, 2020-08-30:
If you prefer to stick with python my fork implements openweathermap. Just put your owm  key on config and follow the standard install directions.

n0bel, 2020-08-31:
During a discussion in #179, the choice of a new provider was decided.  Since then I've been testing that and a few others.   That was 2 months ago.   My available time has been limited.  I've been busy with work, more so than ever before.   While that's good news in one respect, my other projects suffer of course.   I was also hoping that one of the great followers of this project might take on the changes.   This hasn't happened either.   Since I have many of these PiClocks running for my friends and family, I will be forced to update it before the API expires in 2021.    So, no, its not dead, but progress has been slow.

Knackster, 2020-08-31:
Good morning Kevin, and the rest of the great peeps following. I wish I could do more, but alas, I am not a developer. I just wanted to state that I appreciate the time that you and others have taken to get the new solution going.

feh123, 2020-08-31:
I fully agree. Thanks @n0bel for all your work.

BertLindeman, 2020-08-31:
Yes, the PiClock is a fixed asset in my LEGO room ;-)

@n0bel I am in no way a solid programmer (well solid al right, but . . . )
I am willing to contribute but it would be wise to hand me small parts at a time. 
If there are tasks like that, you can always ask and indicate me how/where!
So I could create pul requset(s).

Thanks for the  so flexible PiClock.

captain, 2020-09-05:
> If your brave: https://github.com/merky1/PiClock
> 
> 
> 
> If you have the PiClock up and running today, I’d say stay until the darksky api turns off.  Hopefully there’s enough smart people here to fix that issue.

I Apparently got here a month too late, and didn't get a Dark Sky API key￼￼. What's the trick to getting yours running?￼

indymx, 2020-09-05:
You don't.

N4IK, 2020-09-05:
Get the N4IK fork I created and get an openweathermap key until the main tree is updated correctly.SteveSent from my Verizon, Samsung Galaxy smartphone

merky1, 2020-09-07:
@captain The trick is my fork is not “compatible” with the original, and I have consistently failed to document the settings file, so it takes a higher level of thought right now, being a niche hobby type project.  Being stuck at home hasn’t helped.

Essentially, wipe your pi with the latest raspian, and the follow the instructions in the install.md.  The settings.json is pretty straightforward, but there is no documentation around it.

If you want to stay on the python path, then @N4IK is probably the way to go for now.  My way leads to ; hades...

captain, 2020-09-07:
Thanks y’all!  I got N4IK/PiClock running, figured out how to get weather reports from openweather, but somehow broke python’s ability to use w1thermsensor, which wouldn’tinstall via git but installed fine via apt-get (python -i “import w1thermsensor” claims there is no such file) and I still cannot get dates to display in ISO8601 International Standard format, even when setting locale in Config.py to lt-LT. 😝. I also seem to be missing moon phase information, which is kind of sad. Great work though!  Thanks for the ride. I learned a lot. 😄

N4IK, 2020-09-07:
Openweathermap doesn't have all the same info as dark sky, I just got what I wanted working, hence my previous "quick and dirty" comment. Glad you got it working!SteveSent from my Verizon, Samsung Galaxy smartphone

captain, 2020-09-09:
Does the inside thermometer work on yours n0bel?  I’ve determined that in TempServer.py, python3 executes “from w1thermsensor import W1ThermSensor” correctly, but then doesn’t work with some of the other calls, while python works with the other calls, but doesn’t parse “from w1thermsensor import W1ThermSensor” correctly. 🤪 (apparently python3 requires quotations around parentheticals statements to print) 

I’m really not sure why we are even getting the temperature through this crazy convoluted method of running a Web server just to pass a single number from /sys/bus/w1/devices/[thermometer device number]/w1_slave 
I would just tail -1 that file, then pipe it through cut -c 30- and have a simple number representing degrees Celsius. I’m probably overlooking something here though. 

I have an idea for a cool feature, but I’m barely a hacker, not much of a programmer, so it may take me a while. Here’s my idea, in case anyone else wants to tackle it with me: change the close-up weather window to display video from your front door camera!  I was hoping to use the eufy api, but the bastards at Anker locked it down this summer with “privacy enhancements“. I’m Looking into finding a way to access the video stream from a Samsung V6431BN.

N4IK, 2020-10-09:
Sorry Captain, I just now happened to see this. I don't have an inside temperature sensor so I don't know if it would work or not.

Prestonh996, 2020-11-05:
Any new updates to the instructions for the installation of the new weather API provider?
Thanks!

merky1, 2020-11-09:
> Any new updates to the instructions for the installation of the new weather API provider?
> Thanks!

short answer is there are several forks, but Nobel has IRL things taking priority:

https://github.com/n0bel/PiClock/issues/185#issuecomment-683778413

So the options are to dive into a fork, or wait it out.

Prestonh996, 2020-11-09:
Thanks for the heads up Merky1.

feh123, 2020-12-12:
Hi - I did find this review: https://rapidapi.com/blog/access-global-weather-data-with-these-weather-apis/.

I have not checked it out in detail. My coding skills do not allow me to do anything more useful I am afraid!

N4IK, 2020-12-14:
feh123, n0bel has chosen one but just hasn't had the time to program the changes. Feel free to try my fork where I threw together a temporary openweathermap solution.

feh123, 2020-12-14:
Apologies - on re-reading my post it now sounds like I was pushing. It was more for info and to say I was sorry I could not help much. I am very happy to wait.

n0bel, 2020-12-23:
For those still using the original PiClock, I'm using my christmas downtime to jump into recoding for Climacell.     Get your API keys here:  https://www.climacell.co/weather-api/    I'm doing a quick and dirty Christmas Update to the PiClock code.   Then a new years update for the build from scratch instructions including updates for new versions of Raspberry Pi OS.

feh123, 2020-12-24:
@n0bel Thanks a lot! PiClock is brilliant.

n0bel, 2020-12-24:
Merry Christmas.!  ClimaCell.co is implemented in PiClock.  Bonus!  OpenWeatherMap is also an option.    Remember you still have another year to upgrade from DarkSky.net (if you have keys)

Update instructions are here: https://github.com/n0bel/PiClock/blob/master/Documentation/Install.md#updating-to-newerupdated-versions

Extra Notes:
ApiKeys.py should only have one (1) of the following keys defined:
ccapi   for ClimaCell.co
owmapi  for OpenWeatherMap.org
dsapi   for DarkSky.net  (yes it still works)

Known Issues:
ClimaCell is lacking language specific translations for the current conditions (Rain, Snow, Cloudy, etc)

n0bel, 2020-12-25:
I pushed a few fixes for those who've updated already.  Need to pull again.

n0bel, 2020-12-25:
Another fix was pushed..   You'll want to pull again.   Day/night icons were corrected.

n0bel, 2020-12-26:
Be sure you get a Climacell V3 apikey through this link: V3 Keys can still be obtained from here:  https://developer.climacell.co/sign-up

Other paths lead to a V4 apikey, and I have found issues with the V4 API.

ScottChapman, 2020-12-26:
> Be sure you get a Climacell V3 apikey through this link: V3 Keys can still be obtained from here: https://developer.climacell.co/sign-up
> 
> Other paths lead to a V4 apikey, and I have found issues with the V4 API. #195

Uh, wait. that brings me to a sign up page. Do we need to re-register in order to get a V3 key?

n0bel, 2020-12-26:
> Uh, wait. that brings me to a sign up page. Do we need to re-register in order to get a V3 key?
Yes.  You will have two separate ways to login as well:
V3: https://developer.climacell.co/sign-in
V4: https://app.climacell.co/signin

ScottChapman, 2020-12-26:
>  Yes.  You will have two separate ways to login as well:
>  V3: https://developer.climacell.co/sign-in
>  V4: https://app.climacell.co/signin

Oh, huh. I notice that when I go to their main page (https://www.climacell.co) there is a signing link with two options; weather API and ClimaCell Platform. SO, I'm assuming if we go through via the weather API login, THAT key is right?

n0bel, 2020-12-26:
Yes.  As far as I can tell, Weather API Login is V3.    ClimaCell platform is V4

n0bel, 2020-12-27:
@merky1 what do you use for the current conditions when using NOAA?   I'm investigating api.weather.gov to add it as yet-another-weather-api option within PiClock.

merky1, 2020-12-27:
NOAA as a provider is problematic.  First, you will need moon and sun phase/time generators.  Second, it has a bad habit of returning null values, and updates once per hour at best.  The level of work needed to protect the app against NOAA failures is too great, and I would not recommend it for current.

Also, the API requires a double tap, once to get a location / station ID, and a second to get the current data.

Essentially, there is a large amount of work needed to make NOAA stable, and no guarantee of value.  I added it due to boredom / curiosity, and was underwhelmed by the broken.  On the plus side it forced me to do better error handling 👍

merky1, 2020-12-27:
Spam post, but here’s the log of a station in HI that’s reliably broken:

2020-12-27 11:55:18 AM -0500 info: initLoop
2020-12-27 11:55:18 AM -0500 info: https://api.weather.gov/points/19.924,-155.8875
2020-12-27 11:55:18 AM -0500 info: PiClock listening on port 8081 in server mode
2020-12-27 11:55:19 AM -0500 info: update cur provider nws
2020-12-27 11:55:19 AM -0500 warn: current gridpoint data not updated
2020-12-27 11:55:19 AM -0500 warn: set next current poll to : 1609088059486
2020-12-27 11:55:19 AM -0500 info: update NWS alerts
2020-12-27 11:55:19 AM -0500 info: https://api.weather.gov/alerts/active?point=19.924,-155.8875
2020-12-27 11:55:19 AM -0500 info: update forecast provider
2020-12-27 11:55:19 AM -0500 info: https://api.usno.navy.mil/rstt/oneday?date=now&coords=19.924,-155.8875
2020-12-27 11:55:19 AM -0500 warn: forecast gridpoint data not updated
2020-12-27 11:55:19 AM -0500 warn: set next forecast poll to : Sun Dec 27 2020 10:56:19 GMT-0500 (EST)
2020-12-27 11:55:21 AM -0500 error: getaddrinfo ENOTFOUND api.usno.navy.mil api.usno.navy.mil:443
2020-12-27 11:55:21 AM -0500 warn: generated moon phase with suncalc at 0.42160006713548914
2020-12-27 11:55:21 AM -0500 info: https://api.weather.gov/gridpoints/HFO/236,82/stations
2020-12-27 11:55:21 AM -0500 info: got NWS weather station PHKO
2020-12-27 11:56:30 AM -0500 info: update cur provider nws
2020-12-27 11:56:30 AM -0500 info: https://api.weather.gov/stations/PHKO/observations/latest
2020-12-27 11:56:30 AM -0500 info: update NWS alerts
2020-12-27 11:56:30 AM -0500 info: https://api.weather.gov/alerts/active?point=19.924,-155.8875
2020-12-27 11:56:30 AM -0500 info: update forecast provider
2020-12-27 11:56:30 AM -0500 info: https://api.usno.navy.mil/rstt/oneday?date=now&coords=19.924,-155.8875
2020-12-27 11:56:30 AM -0500 info: https://api.weather.gov/gridpoints/HFO/236,82/forecast
2020-12-27 11:56:30 AM -0500 error: getaddrinfo ENOTFOUND api.usno.navy.mil api.usno.navy.mil:443
2020-12-27 11:56:30 AM -0500 warn: generated moon phase with suncalc at 0.42162444804447213
2020-12-27 11:56:30 AM -0500 warn: stale update detected with timestamp : Sun Dec 27 2020 10:53:00 GMT-0500 (EST) behind c
urrent timestamp by : -26818073 behind now by : 64 minutes
2020-12-27 11:56:30 AM -0500 error: Cannot read property 'periods' of undefined
2020-12-27 11:56:30 AM -0500 warn: set next forecast poll to : Sun Dec 27 2020 10:57:30 GMT-0500 (EST) current time stamp 
322
2020-12-27 11:57:31 AM -0500 info: update NWS alerts
2020-12-27 11:57:31 AM -0500 info: https://api.weather.gov/alerts/active?point=19.924,-155.8875
2020-12-27 11:57:31 AM -0500 info: update forecast provider
2020-12-27 11:57:31 AM -0500 info: https://api.usno.navy.mil/rstt/oneday?date=now&coords=19.924,-155.8875
2020-12-27 11:57:31 AM -0500 info: https://api.weather.gov/gridpoints/HFO/236,82/forecast
2020-12-27 11:57:31 AM -0500 error: getaddrinfo ENOTFOUND api.usno.navy.mil api.usno.navy.mil:443
2020-12-27 11:57:31 AM -0500 warn: generated moon phase with suncalc at 0.4216458455096096
2020-12-27 11:58:30 AM -0500 info: update cur provider nws

[image]

Techfun49, 2020-12-27:
Many thanks to n0bel and others who have helped make this update happen !!  It is greatly appreciated.

With your instructions, the update was successfully implemented.  However, I did notice that the moon phase is no longer displayed at the bottom of the screen.  Is this a function of ClimaCell not providing the info or an issue with the update?

Thanks again!

n0bel, 2020-12-27:
At this point openweathermap seems to be a viable (although erratic/delayed) option.  OWM remain a permanent part of PiClock.   I think I'll spend the time over the next few days experimenting with nws.

n0bel, 2020-12-27:
Moonphase is available,  I forgot to implement it.
However I'm working on changing sunrise/sunset/moonphase in PiClock directly.

merky1, 2020-12-27:
I take it, you can override the NWS nulls by parsing the METAR directly.

PHKO 271853Z 15006KT 10SM CLR 26/17 A3007 RMK AO2 SLP180 T02610167

Winds at 06knots from 150degrees.  Temp is 26.1 with a dew point @16.7 (Celsius).   Parsing rules loosely @ https://en.wikipedia.org/wiki/METAR#Information_contained_in_a_METAR

n0bel, 2020-12-27:
Yes that's an option too.   I haven't started yet.

ScottChapman, 2020-12-28:
still lurking. Does it mostly work at this point? I assume it would be worth pulling the trigger on the update (even if there are minor tweaks coming).

n0bel, 2020-12-28:
Remember the DS API will keep working until the end of 2021.   So you have time.   Things mostly work, Yes, with a V3 key only.  V4 keys will have errors, since PiClock is using V3 until the V4 data issues are fixed by ClimaCell.    The current PiClock/master now supports openweathermap as well, and as well still supports DS.

ScottChapman, 2020-12-28:
Is the way to switch back-and-forth simply to specify the right key in `ApiKeys.py`?

ScottChapman, 2020-12-28:
updated, looks great. Not sure why I am always a tad nervous when I update; always goes well!

ScottChapman, 2020-12-28:
Looks like [V3 will stop working](https://developer.climacell.co/v3/docs/deprecation-notice) mid-year

n0bel, 2020-12-28:
Yup..  Hopefully V4 will be working correctly by then.

n0bel, 2020-12-28:
Back and forth is simply by only defining one API key.
One of:
dsapi
owmapi
ccapi

N4IK, 2020-12-31:
Just switched over today with ClimaCell. Looks good.

n0bel, 2021-01-02:
Just added the METAR data as an option to replace current conditions.
```
# Optional Current conditions replaced with observations from a METAR station
# METAR is world wide, provided mostly for pilots
# But data can be sparse outside US and Europe
#  If you're close to an international airport, you should find soemthing close
# Find the closest METAR station with the following URL
# https://www.aviationweather.gov/metar
# scroll/zoom the map to find your closest station
# or look up the ICAO code here:
# https://airportcodes.aero/name
METAR = ''
```

ScottChapman, 2021-01-04:
So, can someone describe (someplace) how METAR replaces or supplements the info coming from the weather provider?

[99 more replies]