n0bel/PiClock #190: Ghosting on radar

Commander-Crash, 2020-12-23:
I don't know if this is a PiClock issue or Radar issue with the API, I am using PiClock on a atomic pi running MX linux 19. I have notice radar glitches and ghosting whear storm passes and it will leave a trail or particals and what looks like lines show up. I do hope this project is still alive I really like this Program.

n0bel, 2020-12-23:
I've seen this before.  It appears to come from the provider.   Altho in the past I have had issues with certain types of properly formed png files that python or pyqt didn't process correctly.   If you'd like to check yourself, take a look at the log files.  They have the full URL of each radar maptile.  You can copy that URL to your browser, or download with wget/curl and see if you see the problem in the original data.

closed 2020-12-24 by n0bel

SerBrynden, 2023-04-16:
See https://github.com/n0bel/PiClock/issues/241#issuecomment-1510003323