n0bel/PiClock #191: Climacell PiClock api implementaion lacks Language specific weather descriptions

n0bel, 2020-12-24:
Climacell PiClock api implementaion lacks Language specific weather descriptions, i.e. Snow, Rain, Cloudy, etc

n0bel, 2020-12-25:
Added language specific weather condition terms as an optional config item  Lcc_code_map.   Config-Example and Config-Example-Berlin shows some examples.    

Please correct my Google Translated version, and let me know what the proper words are.

https://github.com/n0bel/PiClock/commit/53b696ec7956bc61943275168f629c8475e3d398

BertLindeman, 2020-12-26:
My native language in Dutch, but a bit German I know.
Looked] at the examples and to me it looks OK. Though a real German look at it is preferred ;-)
[Some German weather terms](https://www.thoughtco.com/english-german-glossary-weather-words-4085512)
as hints.

closed 2021-01-04 by n0bel