n0bel/PiClock #192: Climacell PiClock api implementation daily weather descriptions very terse

n0bel, 2020-12-24:
Would like to make a more descriptive weather description for the forecast based on the quantitative data for the period in question.  DarkSky did it for us in the 'summary' field.

merky1, 2020-12-24:
I have been debating adding openweather forecasts.  Currently I use NOAA, and it simplified the process with description and summary being defined.  It’s been a low priority for me, since my fork is US centric.  NOAA also modifies the date with Federal holidays so bonus :)

Climacell seems good only for current observations, and we need another provider for forecasts.

n0bel, 2020-12-25:
ClimaCell seems to work fine for forecasts.  They just don't provide a summary description like Darksky did.    Openweathermap provides a description, but it is a reflection of the weather icon. 

I've captured some weather summaries from dark sky and compared them to the quantitative data.   It looks quite obviously programmatically generated.   With a little thought I think an algorithm could be created.   Certainly I would not attempt a NOAA style weather synopsis for the day.

ScottChapman, 2020-12-25:
ya, I think that also gives you more control over the display; right now (at least for me) it is likely that the text gets too long and truncates at some point which makes it not as great.

closed 2023-04-02 by n0bel