n0bel/PiClock #194: Climacell PiClock KeyErrors

BertLindeman, 2020-12-25:
Took a new clone from git
Updated the config and api
Only 1 key in ApiKeys.py: ccapi for ClimaCell.co

updated python-dateutil and tzlocal 

Now get KeyErrors in the log. 
Added a few prints in the hope that is makes clear why.
Current log:

```
('DEBUG 768 wxdata2:\n', {u'message': u'You cannot consume this service'})
Traceback (most recent call last):
  File "PyQtPiClock.py", line 772, in wxfinished_cc2
    f = wxdata2[i * 3 + 2]
KeyError: 2
('DEBUG 683 tzlocal.get_localzone():', <DstTzInfo 'Europe/Amsterdam' LMT+0:20:00 STD>)
Traceback (most recent call last):
  File "PyQtPiClock.py", line 685, in wxfinished_cc
    f['observation_time']['value'])
KeyError: 'observation_time'
('DEBUG 830 wxdata3:', {u'message': u'You cannot consume this service'})
Traceback (most recent call last):
  File "PyQtPiClock.py", line 835, in wxfinished_cc3
    wxdata3[0]['observation_time']['value']+"T00:00:00Z")
KeyError: 0
```
The line numbers are the original line numbers where the error was reported, 

Switched back to DarkSky for a while and there the KeyErrors did **not** occur,

Any idea where I should look for to further debug this?

n0bel, 2020-12-25:
"You cannot consume this service"  is caused by a problem with your apikey or with the api.   The log file should show the full URL that is being used.  Check that URL in your browser.   Double check the key.

n0bel, 2020-12-25:
I've also done an update for some day/night issues.  You should pull it again.

BertLindeman, 2020-12-25:
> 
> 
> I've also done an update for some day/night issues. You should pull it again.

Thank you.

Will do (both) and report back

ScottChapman, 2020-12-25:
merry xmas?

Knackster, 2020-12-25:
Merry Christmas and Happy New Year

BertLindeman, 2020-12-25:
Sorry @ScottChapman Merry Xmas and a safe, healthy, happy and joyable 2021 to all

[image]

BertLindeman, 2020-12-25:
Something in the sign-up procedure did not go well.
Climacell support is working on it (Even on Xmas) Good support I'd say.
Closing as this is not a PiClock issue.

> 
> 
> merry xmas?

BertLindeman, 2020-12-25:
The api-key I got is a V4 key and that does not work on V3.
And support states that you cannot get a V3 key if you have a V4 key.

Support at Climacell writes to me:
`we are deprecating v3 in a few weeks. V4 is the way to go.`

`
The call you are making is for V3.
this page should should you how to format the V4 call. it is a bit different:

https://docs.climacell.co/reference/get-timelines
`

Should I enter a new issue: Convert Climacell V3 calls into V4?

BertLindeman, 2020-12-26:
Issue #195 has been made, so re-close this one

closed 2020-12-26 by BertLindeman