n0bel/PiClock #195: Migrate to Climacell v4 api

n0bel, 2020-12-26:
Migrate to Climacell v4 api

BertLindeman, 2020-12-26:
URL: [Climacell V4 Api reference](https://docs.climacell.co/reference/)

n0bel, 2020-12-26:
I'm recommending staying with the V3 API for now.  I'm seeing a bug with inaccurate weatherCode fields.   V3 reports snow of various types on each of several days.   V4 reports drizzle and rain, with no snow.   I think the snow codes are not working.

V3 Keys can still be obtained from here:  https://developer.climacell.co/sign-up

n0bel, 2020-12-26:
I've put in a bug report.

BertLindeman, 2020-12-26:
> 
> 
> I'm recommending staying with the V3 API for now. I'm seeing a bug with inaccurate weatherCode fields. V3 reports snow of various types on each of several days. V4 reports drizzle and rain, with no snow. I think the snow codes are not working.
> 
> V3 Keys can still be obtained from here: https://developer.climacell.co/sign-up
> 
> Sign in is separate too: V3: https://developer.climacell.co/sign-in

I can confirm, I now also have a V3 key, THANK you!

merky1, 2020-12-27:
Yeah, the v4 api is significantly lacking.  They haven’t implemented current observations yet.  Their putting v3 o solo team in 2021, but I’m not pulling anything from it until they fix current observations.

feh123, 2020-12-27:
I always wanted to have a display of my diary in the house. Some months ago I found this - https://code.mendhak.com/raspberrypi-epaper-dashboard/. I have had it running now for a few weeks - as a side product of my diary I get a Climacell V3 api weather forecast. So I have seen this working now for some time. It's min/max temp is fine but as I only get the weather-code it's overall weather info is very limited. I found it very useful to edit the weather text to make what I see outside my window fit better with the weather icon displayed. As I am near Cambridge UK I do not see a large variation in weather so cannot say too much their forecasting other than it seems okay. A 970mm depression has just swept over us but weather-code does not forecast this parameter. Thanks @n0bel for all your work on the new code.

ScottChapman, 2021-03-18:
Do we need to try v4 again? https://github.com/n0bel/PiClock/issues/185#issuecomment-798103570
I know for me the V3 forecast doesn't seem great...

merky1, 2021-03-18:
It feels like the v4 api is still missing a concept for realtime.  I pull forecasts from NOAA, so I have no data on their forecast model.

Note

In the upcoming weeks we will introduce another timestep - current - as syntactic sugar returning only this interval per request.

feh123, 2021-04-13:
Hi @ScottChapman I am using V4 and it's forecast accuracy is not good. Currently saying freezing drizzle here when there is nothing but blue sky. Nearly correct on high/low temperature!

merky1, 2021-04-13:
I have noticed that climacells “accuracy” is volatile.  I’ve been pondering yahoo or some other replacement.  Any thoughts on https://www.visualcrossing.com/

ScottChapman, 2021-04-13:
bummer. I rolled back to darksky and it is MUCH better. VisualCrossing does look good though.

mendhak, 2021-04-13:
For what it's worth, I was seeing similar issues with Climacell on [my project](https://github.com/mendhak/waveshare-epaper-display) and have been looking at other providers.  I've just [implemented VisualCrossing](https://github.com/mendhak/waveshare-epaper-display/blob/master/weather_providers/visualcrossing.py) - there's a timeline, it's pretty straightforward to work with and descriptions are clear.  One thing to look out for though, [the icon set is pretty limited](https://www.visualcrossing.com/resources/documentation/weather-api/defining-icon-set-in-the-weather-api/).

merky1, 2021-05-02:
So with testing, both visualcrossing and weatherapi have issues.

Visual crossing only seems to update every 20-30 minutes.
weatherapi seems to update every 15 minutes, but there is a missed collection at the top of every hour.

Both seem to be more accurate than climacell, but have a much lower update rate.  So you lose 10-20% accuracy to keep climacell update rate.  I’m going to keep looking around, I may even consider accuweather *shudder*.

n0bel, 2021-05-02:
I appreciate all the effort.

ScottChapman, 2021-06-08:
have you looked into tomorrow.io?

merky1, 2021-06-08:
It’s the same as climacell, with about 10-20% data discrepancies.  There atmospheric pressure seems to be the worst.

2021-05-30 07:48:57 PM -0400 info: darksky : 53.81 : 0 : Overcast
2021-05-30 07:48:57 PM -0400 info: visualcrossing : 54.1 : null
2021-05-30 07:48:57 PM -0400 info: climacell : 52.97 : Partly Cloudy
2021-05-30 07:48:57 PM -0400 info: openweather : 53.31 : 0 : Clouds
2021-05-30 07:48:57 PM -0400 info: weatherapi : 55.9 : Overcast
2021-05-30 07:48:59 PM -0400 info: usg : 50 F : timestamp age = 14
2021-05-30 07:58:56 PM -0400 info: visualcrossing : 54.1 : null
2021-05-30 07:58:56 PM -0400 info: darksky : 53.77 : 0 : Overcast
2021-05-30 07:58:56 PM -0400 info: climacell : 53.02 : Cloudy
2021-05-30 07:58:56 PM -0400 info: openweather : 53.2 : 0 : Clouds
2021-05-30 07:58:57 PM -0400 info: weatherapi : 55.9 : Overcast
2021-05-30 07:58:58 PM -0400 info: usg : 50 F : timestamp age = 24
apitest-2021-05-30.log lines 1-32/32 (END)

merky1, 2021-06-08:
More log spam.  Tomorrow.io has the "most up to date readings" but the values are all over the map, and the pressure values are not confirmable anywhere else.  Visualcrossing seems to be the most accurate, but updates lag 15-30 minutes.  Openweather is just to variable(timestamps) to trust the data from.  Weather API seems to put a 20 minute hole at every hour, so I moved off of it quickly.

2021-06-08 03:47:31 PM -0400 info: darksky : 89.29 : 0 : Partly Cloudy
2021-06-08 03:47:31 PM -0400 info: climacell : 90.75 : Light Rain
2021-06-08 03:47:31 PM -0400 info: visualcrossing : 90.1 : Partially cloudy
2021-06-08 03:47:31 PM -0400 info: weatherapi : 90 : Partly cloudy
2021-06-08 03:47:31 PM -0400 info: openweather : 88.45 : 0 : Rain
2021-06-08 03:47:31 PM -0400 info: usg : NaN F : timestamp age = 33 : Partly Cloudy
2021-06-08 03:50:02 PM -0400 info: usg : 86 F : timestamp age = 35 : Partly Cloudy
2021-06-08 03:50:02 PM -0400 info: visualcrossing : 90.1 : Partially cloudy
2021-06-08 03:50:02 PM -0400 info: climacell : 90.77 : Light Rain
2021-06-08 03:50:02 PM -0400 info: darksky : 89.31 : 0 : Partly Cloudy
2021-06-08 03:50:03 PM -0400 info: weatherapi : 90 : Partly cloudy
2021-06-08 03:50:09 PM -0400 info: openweather : 88.45 : 3 : Rain
2021-06-08 04:00:02 PM -0400 info: visualcrossing : 90.1 : Partially cloudy
2021-06-08 04:00:02 PM -0400 info: climacell : 90.79 : Light Rain
2021-06-08 04:00:02 PM -0400 info: darksky : 89.36 : 0 : Partly Cloudy
2021-06-08 04:00:02 PM -0400 info: weatherapi : 90 : Partly cloudy
2021-06-08 04:00:03 PM -0400 info: openweather : 88.52 : 0 : Rain
2021-06-08 04:10:02 PM -0400 info: visualcrossing : 88.6 : Partially cloudy
2021-06-08 04:10:02 PM -0400 info: weatherapi : 90 : Partly cloudy
2021-06-08 04:10:02 PM -0400 info: darksky : 89.34 : 0 : Partly Cloudy
2021-06-08 04:10:02 PM -0400 info: climacell : 90.79 : Light Rain
2021-06-08 04:10:03 PM -0400 info: openweather : 88.47 : 0 : Rain
2021-06-08 04:20:02 PM -0400 info: climacell : 90.81 : Light Rain
2021-06-08 04:20:02 PM -0400 info: darksky : 89.27 : 0 : Partly Cloudy
2021-06-08 04:20:02 PM -0400 info: visualcrossing : 88.4 : Clear
2021-06-08 04:20:03 PM -0400 info: weatherapi : 90 : Partly cloudy
2021-06-08 04:20:04 PM -0400 info: openweather : 88.5 : 0 : Rain
2021-06-08 04:30:02 PM -0400 info: visualcrossing : 88.4 : Clear
2021-06-08 04:30:02 PM -0400 info: darksky : 89.12 : 0 : Partly Cloudy
2021-06-08 04:30:02 PM -0400 info: climacell : 90.84 : Light Rain
2021-06-08 04:30:03 PM -0400 info: weatherapi : 90 : Partly cloudy
2021-06-08 04:30:04 PM -0400 info: openweather : 88.48 : 0 : Rain

closed 2021-12-09 by n0bel