n0bel/PiClock #196: Open weather clock does not show 'wind', 'feels_like' and 'timestamp' but to missing wind_gust

BertLindeman, 2020-12-28:
When  using the open weather map key, the json returned does not contain `wind_gusts`.

The log reports that as 
```
KeyError: 'wind_gust'
```

Do you want me to make a pull request to change this 
``` 
        wind.setText(Config.LWind +
                     wd + ' ' +
                     '%.1f' % (speedm(f['wind_deg'])) + 'kmh' +
                     Config.Lgusting +
                     '%.1f' % (speedm(f['wind_gust'])) + 'kmh')
```

into this (and similar for non-metric)

```
        try:
            wind.setText(Config.LWind +
                     wd + ' ' +
                     '%.1f' % (speedm(f['wind_deg'])) + 'kmh' +
                     Config.Lgusting +
                     '%.1f' % (speedm(f['wind_gust'])) + 'kmh')
        except:  # No wind_gust available on open  weather map
            wind.setText(Config.LWind +
                     wd + ' ' +
                     '%.1f' % (speedm(f['wind_deg'])) + 'kmh')
```

Not sure:

- if this is a correct way to program this.
- what to do with my pull requests here, probably useless now ;-)

BertLindeman, 2020-12-29:
Ugh... 
There is a copy/paste error in this area. **Also in the original source**
We show `wind_deg` as if it was `wind_speed`

So I see now a wind_speed of 130 kph (~ 80 mile/hour)
And there is hardly a breeze ;-)

feh123, 2020-12-29:
My upside is that weather underground and dark sky were both very good and hence were bought. I think this is their strategy - be good and get a lot of money when sold. So Climacell will be the same - hence V4? Even OWM - not sure but if MS can buy Github who knows?

n0bel, 2021-01-01:
Commit https://github.com/n0bel/PiClock/commit/6491c57da2ccfa4d6d4c2675ac604c3ee64d8ba1  
Corrected metric wind_speed (from wind_deg) and wind_gust is now optional based on the return values.

BertLindeman, 2021-01-01:
Confirmed working with and without wind_gust

Thanks, @n0bel

closed 2021-01-01 by BertLindeman