n0bel/PiClock #197: I have a DarkSky account and that version of PiClock works. How can Put this version from my SD card to my Hard Drive.

lstew452a, 2020-12-29:
I still have a DarkSky account. How can I get the old verion of Pi Clock using DarkSky API. My old SD card is using DarkSky PiClock and it works fine. How can I remove the new version of PiClock from my hard drive. I want to run the old version on my hard drive. Please advise.

BertLindeman, 2020-12-29:
The new (current) version does run just as good as what you call the old version and both support the DarkSky apikey. For as long as DarkSky allows us.

If the config.py or apikeys.py differ, just copy them over from the sd-card.

Bert

n0bel, 2021-01-01:
https://duckduckgo.com/?t=ffab&q=how+to+backup+raspi+sd+card&atb=v221-1&ia=web

BertLindeman, 2021-01-03:
@lstew452a 
Can this issue be closed?

closed 2021-04-20 by n0bel