n0bel/PiClock #200: In functions wxfinished_* has "global wxdesc" but not wxdesc2

BertLindeman, 2021-01-03:
Start of wxfinished_owm:
```
def wxfinished_owm():
    global wxreply, wxdata, supress_current
    global wxicon, temper, wxdesc, press, humidity
    global wind, wind2, wdate, bottom, forecast
    global wxicon2, temper2, wxdesc, attribution
```
Assume the last line should have `wxdesc2` in stead of `wxdesc`

The same/similar goes for 
```
def wxfinished_ds():
def wxfinished_cc():
def wxfinished_metar():
```

Or if I am wrong, the duplicate global for wxdesc should be removed.

n0bel, 2021-01-03:
I'm not sure why things worked.. but what you found is a bug.

Corrected on https://github.com/n0bel/PiClock/commit/fcc81e8d08868492e8cf7cf519e83bc2f952b746

BertLindeman, 2021-01-03:
**YES**
I also wondered but my python is not rock-solid

closed 2021-01-03 by BertLindeman