n0bel/PiClock #201: PIClock stopped working.

lstew452a, 2021-01-03:
When I went back to using DarkSky for PIClock every started to work fine. But a few days ago PIClock stopped working. How do find out what is wrong.

I found out I am an error 

------------------
(program exited with code: 127)

BertLindeman, 2021-01-03:
Any error seen in the log file?

If not, it might help if you describe your system / how the PiClock is started,

lstew452a, 2021-01-03:
> Any error seen in the log file?
> 
> If not, it might help if you describe your system / how the PiClock is started,
It says PiClock is starting but it does not come up.

lstew452a, 2021-01-03:
> > Any error seen in the log file?
> > If not, it might help if you describe your system / how the PiClock is started,
> > It says PiClock is starting but it does not come up.

Raspbrry PI 400

BertLindeman, 2021-01-03:
> 
> Any error seen in the log file?
> 

The logfile(s) are written to:
```
/home/pi/PiClock/Clock/PyQtPiClock.1.log
/home/pi/PiClock/Clock/PyQtPiClock.2.log
```
and so on. log 1 should be the most recent.

Are these files recent? 
maybe issue on a terminal on the pi400:
`ls -l /home/pi/PiClock/Clock/PyQtPiClock.*.log`

You could look into the most recent log, e.g. using in the terminal:

`nano /home/pi/PiClock/Clock/PyQtPiClock.1.log`

and look  for any python error.

**other posibility**
Did you do any software upgrade since the last time PiClock ran OKE?

lstew452a, 2021-01-03:
> nano /home/pi/PiClock/Clock/PyQtPiClock.1.log

Here is what the log says:

 File "PyQtPiClock.py", line 409
    print "getting current and forecast:" + time.ctime()
                                        ^
SyntaxError: invalid syntax

I did not make any change.
I wonder if Dark Sky turned off all trail accounts.
All Dark Sky activity stopped 1/2.

I looked up the error. It is DarkSky.
Here is the code.

def getwx():
    global wxurl
    global wxreply
    print "getting current and forecast:" + time.ctime()
    wxurl = 'https://api.darksky.net/forecast/' + \
        ApiKeys.dsapi + \
        '/'
    wxurl += str(Config.location.lat) + ',' + \
        str(Config.location.lng)
    wxurl += '?units=us&lang=' + Config.Language.lower()
    wxurl += '&r=' + str(random.random()

ScottChapman, 2021-01-03:
That doesn't look like the [current file](https://github.com/n0bel/PiClock/blob/master/Clock/PyQtPiClock.py#L409)

BertLindeman, 2021-01-03:
Just verified that Dark Sky still works:
[image]

BertLindeman, 2021-01-03:
Is the default python on your Pi400 maybe **python3**?

PiClock only runs python2.

Maybe there has been some change on your Pi with regard to **python**

BertLindeman, 2021-01-03:
In a terminal:
```
python -V
```
shows the version

n0bel, 2021-01-03:
That is exactly what is wrong.   python3 is the default python on whatever system you're using.

Quick fix:  in startup.sh change occurrences python to python2.  There may be some python libraries missing as well.

lstew452a, 2021-01-03:
> That is exactly what is wrong. python3 is the default python on whatever system you're using.
> 
> Quick fix: in startup.sh change occurrences python to python2. There may be some python libraries missing as well.

I ran a tests and here are the results SD card vs Hard Drive on a Raspberry Pi 400 and PiClock
1. Both are using Rasoberry PI OS(32-bit) released 2020-12-02
2. Both use pi@raspberrypi:~/PiClock $ python -V
Python 2.7.16
3. PiClock install steps are the same.
4. Config files are the same.
5. When I boot from the SD card PiClock works fine.
6. When I boot from the Hard Drive PIClock fails with error 
pi@raspberrypi:~/PiClock $ sh startup.sh -n -s
Disabling screen blanking....
Setting sound to max (assuming Monitor Tv controls volume)....
Checking for NeoPixels Ambilight...
Checking for GPIO Buttons...
Starting gpio-keys Service...
Checking for Temperature Sensors...
Starting PiClock.... logging to screen.
ioctl UI_DEV_CREATE: Invalid argument
Traceback (most recent call last):
  File "PyQtPiClock.py", line 8, in <module>
    import dateutil.parser
ImportError: No module named dateutil.parser

BertLindeman, 2021-01-03:
The last line is the problem as you probably already know.
Some of the requirements are missing on that system,
from  the issue [Darksky API Replacement](https://github.com/n0bel/PiClock/issues/185#issuecomment-753545248)

```
Extra Notes:
Updated dateutil and tzlocal may be needed:
sudo pip install python-dateutil --upgrade
sudo pip install tzlocal --upgrade
```

lstew452a, 2021-01-04:
> sudo pip install tzlocal --upgrade
That upgrade fixed it. Thanks.

closed 2021-01-04 by lstew452a