n0bel/PiClock #202: error. no sunrise and sunset data.

xenon462, 2021-01-04:
Error:
Traceback (most recent call last):
  File "PyQtPiClock.py", line 1361, in qtstart
    sunrise = sun.sunrise(dt)
  File "PyQtPiClock.py", line 46, in sunrise
    self.__calc()
  File "PyQtPiClock.py", line 123, in __calc
    math.tan(math.radians(declination)))))
ValueError: math domain error

The polar night is now beyond the Arctic circle.
ClimaCell does not provide sunrise and sunset data.

my coordinates:
68.97033228006596, 33.07604959033541

n0bel, 2021-01-04:
Corrected on commit dd0948439553fdb595ddf6cfdb7dec3d8c4ef319

n0bel, 2021-01-04:
about 24 hours ago I changed the sunrise/sunset computation to be internal to PiClock.  That's where the error was coming from.   I've since corrected it.   Pull and update and you should be fine.

xenon462, 2021-01-04:
updated:
cd PiClock
git pull
python update.py

error appears after update:

Traceback (most recent call last):
  File "PyQtPiClock.py", line 1361, in qtstart
    sunrise = sun.sunrise(dt)
  File "PyQtPiClock.py", line 46, in sunrise
    self.__calc()
  File "PyQtPiClock.py", line 123, in __calc
    math.tan(math.radians(declination)))))
ValueError: math domain error

in previous versions, you fixed a similar bug.
https://github.com/n0bel/PiClock/issues/118

n0bel, 2021-01-04:
The error is exactly the same as before. 
I don't believe the 'git pull' worked, because the line numbers should have changed.
Can you verify that line 127 is as follows:
```
if hourangle0 > 1.0:
```

xenon462, 2021-01-04:
after updating 'git pull' everything worked correctly. the question can be closed.

closed 2021-01-04 by n0bel