n0bel/PiClock #203: Using Berlin example: unsupported operand type(s) for +: 'float' and 'unicode'

BertLindeman, 2021-01-04:
Running RaspiOs with Python 2.7.16
At current `git pull` level

Error while testing Config-Example-Berlin

```
('wxmetar', 'EDDB 041620Z 06009KT 9999 BKN006 01/M00 Q1017 NOSIG')
Traceback (most recent call last):
  File "PyQtPiClock.py", line 1191, in wxfinished_metar
    '%.1f' % (tempm(feels_like(f)) + u'°C'))
TypeError: unsupported operand type(s) for +: 'float' and 'unicode'

```

I do not understand why the lines before this error **are** accepted...
```
        ws = (Config.LWind +
              wd + ' ' +
              '%.1f' % (f.wind_speed.value('KMH')) + 'kmh')
        if f.wind_gust:
            ws += (Config.Lgusting +
                   '%.1f' % (f.wind_gust.value('KMH')) + 'kmh')
```

Ah, those are integers, not float.... OKE.

Will try to make a pull request, including the typo (I assume) in `wxur3` for `wxurl3`

n0bel, 2021-01-04:
Corrected 'Feels Like' for metric  metar  https://github.com/n0bel/PiClock/commit/34ca8b3ad35fcd97e3ba1e9457ba290478a4c4dc

BertLindeman, 2021-01-04:
Confirmed

Thanks Kevin

closed 2021-01-04 by BertLindeman