n0bel/PiClock #206: The top line does not display the days of the week with regional settings.

xenon462, 2021-01-05:
[image]

In the PyQtPiClock.py file I added the line
 
ds = ds.decode('utf-8')
ds2 = ds2.decode('utf-8')

[image]

The top line worked correctly.
[image]

[image]

How do I fix the days of the week in the column on the right?

My settings:
[image]

xenon462, 2021-01-05:
Added "decode ('utf-8')" next to line 909
[image]

The right column only displays the top three rows correctly.
[image]

The other six days of the week are not displayed.

xenon462, 2021-01-05:
Made changes on line 976
section  'wxfinished_cc3'

[image]

All days of the week work
[image]

How do I ask Kevin to make these fixes to the main code of the program?

BertLindeman, 2021-01-05:
The procedure would be to edit [Clock/PyQtPiClock.py](https://github.com/n0bel/PiClock/blob/master/Clock/PyQtPiClock.py)
There is a pencil icon near the top on the right side

Edit your changes 
and you should be able to create a pull request.

And then it is up to Kevin to accept or reject.

It's no rocket science but github can be confusing at times.

xenon462, 2021-01-05:
I can't see a pencil. I will search now. Thanks for the help.

ScottChapman, 2021-01-05:
You sure you don't see this:
[image]
Maybe you aren't logged in?

BertLindeman, 2021-01-05:
@xenon462 has added #207 for this issue

closed 2023-04-02 by n0bel