n0bel/PiClock #210: No internal temperature

briantogo, 2021-01-21:
I am setting up a new RP3, and can get everything but the internal temp to display.  I am using the DS18B20, and have entered the proper device id in tempnames.py, but the internal temp doesn't display on PiClock.  When I run TempServer.py, I get the following error;
Traceback (most recent call last):
  File "TempServer.py", line 7, in <module>
    from w1thermsensor import W1ThermSensor
ImportError: No module named w1thermsensor

I'm reloading Raspberry OS again overnight and will try again in the morning.  Any suggestions would be helpful, thanks.

briantogo, 2021-01-22:
Reloading PiClock on a fresh install of Raspberry OS, I noticed an error when the DS18B20 Temperature driver for Python:  

Traceback (most recent call last):
  File "setup.py", line 3, in <module>
    from pathlib import Path
ImportError: No module named pathlib

xenon462, 2021-01-22:
Try installing using the command:
sudo apt-get install python3-w1thermsensor

Here is the W1ThermSensor installation page:
https://github.com/timofurrer/w1thermsensor/blob/master/README.md

briantogo, 2021-01-22:
Thanks xenon462.  I didn't get any error messages, and running "w1thermsensor ls" shows my sensor, but I still don't see the internal temp displayed.  Any other suggestions?

xenon462, 2021-01-22:
go to w1thermsensor folder as super user
and run setup.py file in python3

sudo su -
cd w1thermsensor
python3 setup.py install

briantogo, 2021-01-24:
OK, so I installed W1ThemSensor as instructed above, but when I try to run setup.py as super user, I get the following error when trying to change directory: cd: w1thermsensor: No such file or directory

So then I installed w1thermsensor using the original PiClock instructions: git clone https://github.com/timofurrer/w1thermsensor.git, when allowed me to browse to the w1thermsensor folder and run "python3 setup.py install" as a super user, and no errors reported.  

The rest of the install went well, but still missing internal temp.  When I run python TempServer.py, I get an error "File "TempServer.py", line 7, in <module>
    from w1thermsensor import W1ThermSensor
ImportError: No module named w1thermsensor"

Any other suggestions would be most appreciated.  Thanks for all the help so far.

xenon462, 2021-01-25:
Here is my setup PiClock
I have Raspberry Pi 2 Model B

I installed Raspbian Jessie
https://downloads.raspberrypi.org/raspbian/images/raspbian-2017-07-05/

I did everything according to the instructions
https://github.com/n0bel/PiClock/blob/master/Documentation/Install-Jessie.md

To install DS18B20 I used the command as root
sudo su -
pip install w1thermsensor

these two lines from the instructions do not need to be executed.
git clone https://github.com/timofurrer/w1thermsensor.git && cd w1thermsensor
python setup.py install

[image]

closed 2021-03-01 by n0bel