n0bel/PiClock #213: File "PyQtPiClock.py", line 64, in __timefromdecimalday     return datetime.time(hour=h, minute=m, second=s) ValueError: hour must be in 0..23

davidabevan, 2021-03-10:
Hello,
Thanks for your project.
I have just started on the Pi moving from arduino.
I have this error upon running and my LCD stops at hat point with a background and unfilled clockface.
I wonder if newer libraries have caused this or is it a silly newbie user error?
this is the error copied from ssh terminal.

Traceback (most recent call last):
  File "PyQtPiClock.py", line 1375, in qtstart
    sunset = sun.sunset(dt)
  File "PyQtPiClock.py", line 54, in sunset
    return suntimes.__timefromdecimalday(self.sunset_t)
  File "PyQtPiClock.py", line 64, in __timefromdecimalday
    return datetime.time(hour=h, minute=m, second=s)
ValueError: hour must be in 0..23
^Z
[1]+  Stopped                 sh startup.sh -n -s

Thanks.. I hope you can advise
Kind Regards 
Dave B

PS I have treble checked my files as per instructions and tried a google etc before posting Its probably me...

n0bel, 2021-04-20:
Is this still an issue?  If so, could you provide your latitude?

davidabevan, 2021-04-20:
Yes I could never solve this.
I live in UK near St Austell Cornwall.
So 50.38001 by -4.795000
I am still learning this stuff 😁
Thanks for replying.

closed 2022-03-06 by n0bel