n0bel/PiClock #215: The Road Forward...

n0bel, 2021-04-20:
I've been waiting for Climacell to get better.   Its not.  I'm running v4 and I'm still not happy.

I've pretty much decided the road forward, but I'll still listen to input.   It seems at this point I'm the only one doing major code changes for the last couple years.

My roadmap looks like this.
Implement ClimaCell v4 (it is already implemented in a branch)
Keep Darksky, OpenWeatherMap (and possibly another API with a pay for option)
Keep the Metar API for current conditions.
Add one or more Personal Weather Station APIs (Ambient Weather will be one) mainly to be used for current conditions.

What does this give us?

The free version of OpenWeatherMap can be 2-3 hours old.   This may be "OK" for the forecast, but is unacceptable for current conditions.   Using METAR for current conditions is a partial solution, but depending on your location, may not be.   Adding PWS APIs can also help solve the Current Conditions issue.

Adding a non-free API (or APIs) to PiClock can also work to get more up-to-date current conditions.

After I've implemented all of that, I'm going to move to a completely new version of PiClock (calling it PiClock3, because I had a false start with PiClock2 a couple years ago).    PiClock3 will have a completely configurable modular format.  Yaml files will contain configuration information, defining pages, frames (parts of pages), and connect any api (or other data source) to any frame.   Each data source (api) will also be implemented in python modules with a standardized interface.  I will be including a conversion script to transform a PiClock configuration to a PiClock3 configuration.

The PiClock3 repository is already created, but very empty, so you can start to follow that project if you wish.

lets see how this goes...

BertLindeman, 2021-04-20:
Very grateful for your dedication, Kevin.

Sure will look forward to PiClock3.

feh123, 2021-04-20:
I agree - thanks a lot.

xenon462, 2021-04-21:
Thanks for your work. 
If you can add the font color editor feature in future versions. 
For example like this. 
[image]
[Config.py.txt](https://github.com/n0bel/PiClock/files/6353315/Config.py.txt)
[PyQtPiClock.py.txt](https://github.com/n0bel/PiClock/files/6353316/PyQtPiClock.py.txt)

maserowik, 2021-04-22:
Would you be able to share you clock images?

Very nice.

 

 <https://user-images.githubusercontent.com/34947093/115607836-22852780-a2ee-11eb-9c3f-cf7775fee2f7.png> 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

xenon462, 2021-04-23:
Here are my watch dial pictures and weather pictures.

[images.zip](https://github.com/n0bel/PiClock/files/6363181/images.zip)
[weather-icons-color.zip](https://github.com/n0bel/PiClock/files/6363182/weather-icons-color.zip)

SerBrynden, 2021-06-29:
climacell.co is now tomorrow.io.  Anyway, I wanted to give v4 a test drive, but when I looked at the code in the climacell v4 branch, it still looks like v3.  Has it not been updated yet?

xxxDrewedxxx, 2021-10-05:
Since I don't need ANOTHER clock on my desk, I have it set to display radar, forecast and moon phases only.  It would be cool if it would pull NWS for weather warnings and display them, but I have another pi that does that
[image]
[image]
.

MoonBoxingRoolRyeRye, 2021-11-03:
> Here are my watch dial pictures and weather pictures.
> 
> [images.zip](https://github.com/n0bel/PiClock/files/6363181/images.zip) [weather-icons-color.zip](https://github.com/n0bel/PiClock/files/6363182/weather-icons-color.zip)

Hey, I love your icons and overall approach to having more colors and whatnot, but I'm having problems with deployment. With the weather icons you've provided, I keep getting "QPixmap::scaled: Pixmap is a null pixmap" for the icons and nothing displays for the weather icon. I am also not able to get the forecast days/times a different color. I'm able to change all of the colors of the days of the week, but I am unable to get them to be different. Any help would be greatly appreciated. I've tried various approaches to get the forecast and the future days to be different, including by default, your approach, but I'm not having any luck. I've attached my main file. I just don't know, I've had to make some modifications here and there to suit my particular use case, but as far as the colors go, it makes sense that it should work in my head, it just doesn't.  Once again, any help would be appreciated. The main changes in my PyQtPiClock include the colors, slideshow randomization and fixing the second page maps for high resolution screens.
[PyQtPiClock.py.txt](https://github.com/n0bel/PiClock/files/7464842/PyQtPiClock.py.txt)

closed 2023-04-02 by n0bel