n0bel/PiClock #217: Blank PI-Clock

maserowik, 2021-05-24:
I like to have my display shut off at night.  I made a new config file and it call correctly but I still get the radar on the screen.  

Here is part of the config that call a blank screen.

location = LatLng(primary_coordinates[0], primary_coordinates[1])
primary_location = LatLng(primary_coordinates[0], primary_coordinates[1])
noaastream = ' '
background = ' '
squares1 = ' '
squares2 = ' '
icons = ' '
textcolor = ' '
clockface = ' '
hourhand = ' '
minhand = ' '
sechand = ' '

here is what i put in the crontab to call the Config-off

0 0 * * * sh /home/pi/PiClock/switcher.sh Config-Off

xenon462, 2021-11-09:
To turn off the display at night, you can add the command to sudo crontab -e

00 22 * * * /usr/bin/vcgencmd display_power 0

0 6 * * * /usr/bin/vcgencmd display_power 1

n0bel, 2021-12-09:
closed after no more activity

closed 2021-12-09 by n0bel