n0bel/PiClock #218: Sunrise and Sunset times are not in Summer time

wlavallee, 2021-06-28:
Hello
I live in Eastern, and the sunrise and sunset times seem to be in Standard time even in DST.  The Time of day is in DST, but not the sunrise/sunset.

Also, the sunrise/sunset times seem to be incorrect.

For example, sunrise this morning was at 5:14AM, but PyClock is saying it is/was at 04:17.

There seems to be a free JSON API for sunrise/sunset at https://api.sunrise-sunset.org/json?lat=XXXX&lng=XXXX

i'd fix it myself, but I don't speak python.  Has anyone else seen this?

wlavallee, 2021-06-28:
[image]

Note the bottom line.

SerBrynden, 2021-06-29:
I noticed the same problem. The sunrise/sunset times are off by an hour and a few minutes.

SerBrynden, 2021-07-01:
@n0bel et al,

in the function `tick()`, look for

`dt = now.replace(tzinfo=tzlocal.get_localzone())`
`sunrise = sun.sunrise(dt)`
`sunset = sun.sunset(dt)`

and change to the following

`dt = datetime.datetime.now(tz=tzlocal.get_localzone())`
`sunrise = sun.sunrise(dt)`
`sunset = sun.sunset(dt)`

In the function `qtstart()`, look for

`dt = datetime.datetime.now().replace(tzinfo=tzlocal.get_localzone())`
`sun = SunTimes(Config.location.lat, Config.location.lng)`
`sunrise = sun.sunrise(dt)`
`sunset = sun.sunset(dt)`

and change to the following

`dt = datetime.datetime.now(tz=tzlocal.get_localzone())`
`sun = SunTimes(Config.location.lat, Config.location.lng)`
`sunrise = sun.sunrise(dt)`
`sunset = sun.sunset(dt)`

For some reason, using `replace(tzinfo=tzlocal.get_localzone())` gives me a timezone of -4:56, but it should be -4:00 for EDT.  This forum post might shed some light on it:
https://stackoverflow.com/questions/60996205/python-datetime-timezone-conversion-off-by-4-minutes

wlavallee, 2021-07-01:
> @n0bel et al,
> 
> In PiClock\Clock\PyQtPiClock.py, in the function `tick()`, look for
> 
> `dt = now.replace(tzinfo=tzlocal.get_localzone())`
> `sunrise = sun.sunrise(dt)`
> `sunset = sun.sunset(dt)`
> 
> and change to the following
> 
> `dt = datetime.datetime.now(tz=tzlocal.get_localzone())`
> `sunrise = sun.sunrise(dt)`
> `sunset = sun.sunset(dt)`
> 
> In the function `qtstart()`, look for
> 
> `dt = datetime.datetime.now().replace(tzinfo=tzlocal.get_localzone())`
> `sun = SunTimes(Config.location.lat, Config.location.lng)`
> `sunrise = sun.sunrise(dt)`
> `sunset = sun.sunset(dt)`
> 
> and change to the following
> 
> `dt = datetime.datetime.now(tz=tzlocal.get_localzone())`
> `sun = SunTimes(Config.location.lat, Config.location.lng)`
> `sunrise = sun.sunrise(dt)`
> `sunset = sun.sunset(dt)`
> 
> For some reason, using `replace(tzinfo=tzlocal.get_localzone())` gives me a timezone of -4:56, but it should be -4:00 for EDT. This forum post might shed some light on it:
> https://stackoverflow.com/questions/60996205/python-datetime-timezone-conversion-off-by-4-minutes

Thanks, the "SunTimes" needs to be changed to "suntimes". after that, it worked AOK!

wlavallee, 2021-07-01:
[image]

wlavallee, 2021-07-01:
Its completely PoE powered, and has an NTP dongle -- so it either keeps up-to-date over NTP, and uses local GPS when available :)

closed 2021-07-01 by wlavallee