n0bel/PiClock #219: API Darksky ending end of 2022

cellfreak, 2021-07-12:
Update: the API service will continue until the end of 2022.

PaPaTheGMan, 2021-07-14:
While the service may continue for existing users, they are not accepting new sign-ups.

open-meteo, 2021-09-22:
Creator of Open-Meteo here

Maybe my humble non-commercial-use weather API could help. I would be happy to assist in the implementation. Let me know if you need any features to make it work easily.

Best, Jeff

PaPaTheGMan, 2021-09-22:
I'd sure appreciate any help you could give.  Since I don't already have a Dark Sky API, PiClock won't work for me.

SerBrynden, 2021-09-22:
> I'd sure appreciate any help you could give. Since I don't already have a Dark Sky API, PiClock won't work for me.

You can currently use PiClock with an OpenWeatherMap API key.

PaPaTheGMan, 2021-09-23:
Thanks, didn't know.  Can you point me to the updated script?  When i download this project, all I get is Dark Sky.

SerBrynden, 2021-09-23:
The default master project at https://github.com/n0bel/PiClock is already setup for OWM.  Just the install instructions haven't been updated.  The https://github.com/n0bel/PiClock/blob/master/Clock/ApiKeys-example.py has the lines for OWM API key.

n0bel, 2021-09-24:
@open-meteo   I'd be happy to implement open-meteo in PiClock.   I'd also solicit help from the community here to help.   I'll open an issue to facilitate getting that task done.     I'll look at the APIs and see how far I get.   Might I ask how often the current conditions are updated?  This is one of the issues with other APIs I've explored, that is current conditions (time now) doesn't get updated often.  So things like "thunderstorm" or "rain" are inaccurate for quick moving current weather conditions.    10 minutes is acceptable, 15 maybe at the worst.    I've resorted to using METAR as an alternative for reporting current conditions, which are updated very quickly, but not available in all areas.

open-meteo, 2021-09-24:
Great to hear!

Current conditions are based on weather models which update every 3 hours in Europe. Currently only 6 hours updates are available in North America, but I rapid updating NOAA models will be added soon. 

Current weather is not based on METAR air port stations. I would love to use real-time measurements to improve current weather predictions, but it will take some time until it is available. In my experience, many weather APIs claim they do nowcasting, radar predictions, realtime satellite integration, etc, etc, but not much is actually visibly in their current weather APIs.

Best, Jeff

n0bel, 2021-12-09:
closing... in favor of #221