n0bel/PiClock #222: Rainviewer not working

feh123, 2021-10-02:
Hi 

As anyone notiiced rainviewer is not working for the UK? If I check the logon messages I see radar1 and radar2 requests to the http site then getFilesReply 0.

I tired a re-boot but still no radar image. Mapbox is working as is Dark Sky.

lupe1337, 2021-10-02:
I am having the same issue in the US.  Everything else is working as intended.

SerBrynden, 2021-10-02:
Most likely another rainviewer outage and not a problem with PiClock.

feh123, 2021-10-02:
Thanks for the fast responses - the rainviewer.com coverage map looks good and nothing to say the api is down. Unfortunately I couldn’t find an obvious way to report an issue.

n0bel, 2021-10-02:
Looks like this is another victim of Lets Encrypt old root cert expiration (sept 30, 2021)

I tried fixing my root certificates on my pi, (outlined below) but that didn't make PiClock work.

So I changed https:// to http:// for rainviewer

Commit to check the diff: https://github.com/n0bel/PiClock/commit/a0938e15e961ca55ca08038ba9e3c8aea2479ccd

line 1624
...
tileurl = "https://tilecache.rainviewer.com/v2/radar/%d/%s" \
...
to
...
tileurl = "http://tilecache.rainviewer.com/v2/radar/%d/%s" \
...

I can't seem to find an automated way to update the root cert bundle on my Pi... so here's what I did..  As stated, this didn't fix PiClock nor cUrl, but it did fix wget.
...
sudo su -
cd /etc/share/ca-certificates
wget --no-check-certificate https://letsencrypt.org/certs/isrgrootx1.pem
echo isrgrootx1.pem >>/etc/ca-certificates.conf
update-ca-certificates --fresh
...

feh123, 2021-10-03:
Hi @n0bel Thanks for this. I tried to use the wget fix but found that my PiClock has no folder /etc/share at all. I have Dark Sky and I have not updated at all yet - is that why I am missing this folder?
However I tried the edit of PyQtPiClock.py altering https to http - that worked and my radar sequence is back! My .py file was also different to yours - the line for the tileurl was not 1624 but around line 600.  
Anyway all looking good now thanks.

n0bel, 2021-12-09:
Updating Raspbian didn't fix the issue.  Its a pyqt4 issue.   
More reason to move to python3 and pyqt5

closed 2021-12-09 by n0bel