n0bel/PiClock #223: No longer getting radar images

ScottChapman, 2021-10-04:
I recently stopped getting radar images; we've got a couple of rainy days and if I go to my local weather app I see greeen radar showing it actually raining (which it is). But my clock shows everything clear (though the forcast shows it at 100% chance of rain).

Any ideas on how I debug this?

I looked in the logs and I see a couple of calls to rainviewer that looks like they succeeded?

xxxDrewedxxx, 2021-10-05:
See Issue #222 
Basically what you need to do is edit PyQtPiClock.py
Look for the line
tileurl = "https://tilecache.rainviewer.com/v2/radar/%d/%s"
And change it to 
tileurl = "http://tilecache.rainviewer.com/v2/radar/%d/%s"

You are only changing https to http, then saving and re-starting.
The line is down quite a way in the code.  Perhaps around line 600?

ScottChapman, 2021-10-05:
Great, thanks. I did an upgrade from Jessie -> Stretch and lost my wifi. So once I get that fixed, I'll kick the tires on that!

It seems that when I did the upgrade I lost wifi, and it no longer appears to be auto-logging in as my user to be able to start up the clock. I figure I'll look at the installation instructions again to resolve that. Are things different in Stretch?

(I will probably go to the next version once I get it working again just to be safe)

closed 2021-10-05 by ScottChapman