n0bel/PiClock #224: upgraded, and clock stopped working (missing tzlocal)

ScottChapman, 2021-10-04:
So, I decided to upgrade my clock to most recent (hasn't been super long). But now I am seeing this in my logs:
```
Traceback (most recent call last):
  File "PyQtPiClock.py", line 9, in <module>
    import tzlocal
ImportError: No module named tzlocal
```
THis is what `pip` shows:
```
pi@raspberrypi:~ $ sudo pip --version
pip 1.5.6 from /usr/lib/python2.7/dist-packages (python 2.7)
pi@raspberrypi:~ $ sudo pip list
argparse (1.2.1)
chardet (2.3.0)
colorama (0.3.2)
gpiozero (1.4.0)
gyp (0.1)
html5lib (0.999)
lxkeymap (0.1)
mcpi (0.1.1)
ndg-httpsclient (0.3.2)
numpy (1.8.2)
picamera (1.13)
pifacecommon (4.2.1)
pifacedigitalio (3.1.0)
Pillow (2.6.1)
pip (1.5.6)
pyasn1 (0.1.7)
pygame (1.9.2a0)
pygobject (3.14.0)
pyOpenSSL (0.13.1)
pyserial (2.6)
python-dateutil (2.8.2)
python-metar (1.4.0)
pytz (2020.5)
requests (2.4.3)
RPi.GPIO (0.6.3)
RTIMULib (7.2.1)
sense-hat (2.2.0)
setuptools (5.5.1)
six (1.16.0)
spidev (3.0)
UNKNOWN (0.0.0)
urllib3 (1.9.1)
wheel (0.24.0)
wsgiref (0.1.2)
pi@raspberrypi:~ $
```

Any ideas at this point?

eeburnham, 2021-10-04:
You need to install tzlocal. Try this:

sudo apt-get install python-tzlocal

ScottChapman, 2021-10-04:
```
pi@raspberrypi:~ $ sudo apt-get install python-tzlocal
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package python-tzlocal
```

ScottChapman, 2021-10-04:
I did run `pip install tzlocal` which appears to result in the `UNKNOWN` entry above

ScottChapman, 2021-10-04:
FWIW - my clock was working fine a couple of weeks ago, I stopped getting radar images overlayed for some reason, so I updated (git pull, then ran update.py)

ScottChapman, 2021-10-04:
ok, I think my problem might be more fundamental. I can't do any `apt-get upgrade` as it is looking for jessie stuff. My sources.list looks like:
```
deb http://mirrordirector.raspbian.org/raspbian/ jessie main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspbian.org/raspbian/ jessie main contrib non-free rpi
```
any thoughts on how to proceed?

ScottChapman, 2021-10-04:
I'm going to upgrade from Jessie -> Stretch and see if that helps

xxxDrewedxxx, 2021-10-05:
See issue 222 adn 223 for the radar problem.

closed 2021-10-05 by ScottChapman