n0bel/PiClock #225: PiClock fails at PyQtPiClock.py", line 1264

lstew452a, 2021-11-11:
This seems to a Dark sky issue.
Please advise.

def getwx_ds():
    global wxurl
    global wxreply
    print "getting current and forecast:" + time.ctime()
    wxurl = 'https://api.darksky.net/forecast/' + \

stylinnile, 2021-11-12:
Same issue here, there a known fix?

n0bel, 2021-11-12:
please show the error message.

n0bel, 2021-11-12:
Also be sure the default python version is version 2, not 3.

lstew452a, 2021-11-14:
> please show the error message.

Setting sound to max (assuming Monitor Tv controls volume)....
Checking for NeoPixels Ambilight...
Checking for GPIO Buttons...
Starting gpio-keys Service...
Checking for Temperature Sensors...
Starting PiClock.... logging to screen.
  File "/home/pi/PiClock/Clock/PyQtPiClock.py", line 1264
    print "getting current and forecast:" + time.ctime()
          ^
SyntaxError: invalid syntax

BertLindeman, 2021-11-14:
n0bel asked: 
> 
> Also be sure the default python version is version 2, not 3.

This error looks like you are running python3 that needs "(" and ")"

How do you start the clock?
And what is your default python version? 
In a terminal:
```python --version```

lstew452a, 2021-11-14:
i change How do the default python --version

BertLindeman, 2021-11-14:
Start on the desktop a terminal session.
Maybe via the application menu -> System tools -> LXterminal.

In the terminal do:
```python --version```

Then copy the result and paste here in a reply.

lstew452a, 2021-11-16:
Support for python 2.7  ended in 2019. Looks like they took it out in latest release of the OS.

stylinnile, 2021-11-16:
How do I add python 2 back into Raspbian? I tried looking up but it came inconclusive

SerBrynden, 2021-11-16:
Try my fork at https://github.com/SerBrynden/PiClock 

I updated the code to Python 3 and PyQt5, and it seems to run well on Raspberry Pi OS versions Buster and Bullseye.  I also added an option to use custom dark maps from MapBox, with a transparent layer for labels, borders, and roads on top of the radar imagery.  This way the map details are obscured by large rain storms.

The changse to Python 3 and Qt5 have worked pretty well, but there are intermittent crashes once in a blue moon that I haven’t fixed yet.  

Occasionally, there are index out of range errors with the radar image tiles.  I suspect the expected number of radar tiles don’t get downloaded properly from RainViewer sometimes.  I’ve also seen JSON decode errors with the OpenWeatherMaps data, segmentation faults, and messages like “python3: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.0.”

I would like to get rid of all the global variables from the original code, but they're prolific.

I haven't tested every feature, such as LED lighting, background switcher, temperature sensor, buttons, or IR remote control, since I don’t use those things.

Rickinfl, 2021-11-16:
Same issue. Bullseye uses PyQt 3.9  and it will not run on it. To be honest it wasn't even worth fighting getting PyQt2 on to it so I just loaded Buster and went about my day.

bl8demast3r, 2021-11-16:
Wow I was just fighting to get pyqt4 working on bullseye and @SerBrynden posted a pyqt5 port. Thanks for that!

lstew452a, 2021-11-17:
I tried using the PIClock and got this error.
isabling screen blanking....
Setting sound to max (assuming Monitor Tv controls volume)....
Checking for NeoPixels Ambilight...
Checking for GPIO Buttons...
Checking for Temperature Sensors...
Starting PiClock.... logging to screen.
Traceback (most recent call last):
  File "/home/pi/PiClock/Clock/PyQtPiClock.py", line 17, in <module>
    from PyQt5 import QtGui, QtCore, QtNetwork, QtWidgets
ModuleNotFoundError: No module named 'PyQt5'

SerBrynden, 2021-11-17:
@lstew452a you need to install PyQt5, per the instructions https://github.com/SerBrynden/PiClock/blob/master/Documentation/Install.md

`sudo apt-get install python3-pyqt5`

lstew452a, 2021-11-17:
> Try my fork at https://github.com/SerBrynden/PiClock 
> 
> I updated the code to Python 3 and PyQt5, and it seems to run well on Raspberry Pi OS versions Buster and Bullseye.  I also added an option to use custom dark maps from MapBox, with a transparent layer for labels, borders, and roads on top of the radar imagery.  This way the map details are not obscured by large rain storms.
> 
> The changes to Python 3 and Qt5 have worked pretty well, but there are intermittent crashes once in a blue moon that I haven’t fixed yet.  
> 
> Occasionally, there are index out of range errors with the radar image tiles.  I suspect the expected number of radar tiles don’t get downloaded properly from RainViewer sometimes.  I’ve also seen JSON decode errors with the OpenWeatherMaps data, segmentation faults, and messages like “python3: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.0.”
> 
> I would like to get rid of all the global variables from the original code too, but they're prolific.
> 
> I haven't tested every feature, such as LED lighting, background switcher, temperature sensor, GPIO buttons, or IR remote control, since I don’t use those things.

Thanks PiClock works again

stylinnile, 2021-11-18:
@SerBrynden Thanks so much it worked first try!

n0bel, 2021-12-09:
I've updated the install documentation to refer to a correct version of Raspberry Pi OS that has Python2 still in it.   The docs always clearly said "For Raspbian Stretch", but but links go to the current download pages, the current version of which has passed PiClock by.    I'll be doing a Python 3 version soon.   An older version ported to Python3 already exists on the dev branch, but it is lacking the newest features.

vanhalen26505, 2021-12-09:
Thanks so much for continuing to make this available! I have it in a bunch
of screens in my house and love it!

Lee

lstew452a, 2022-01-06:
I had to re-install the OS n my raspberry pi and when run PIClock I get error. Traceback (most recent call last):
  File "/home/pi/PiClock/Clock/PyQtPiClock.py", line 15, in <module>
    import dateutil.parser
ModuleNotFoundError: No module named 'dateutil'