n0bel/PiClock #226: Buttons Not Working Under buster

sschnelker007, 2021-12-10:
Has anyone else had a problem with the buttons not working under Buster?  Thx!

n0bel, 2021-12-11:
I'll be testing this, but I'll be targeting PiOS bullseye over the next week or two.

xenon462, 2021-12-11:
Buttons must be connected through pull-up resistors
[image]

sschnelker007, 2021-12-13:
Thanks, but that didn't work either.  I need to read up on how to test if the Pi sees the button being pressed.

sschnelker007, 2021-12-13:
> I'll be testing this, but I'll be targeting PiOS bullseye over the next week or two.

Cool, I will keep an eye on this.  It would be nice to get on the latest OS.  8)

closed 2023-04-02 by n0bel