n0bel/PiClock #228: Retargeting to PiOS/Python3/PyQt5

n0bel, 2021-12-11:
Thanks to @SerBrynden we're moving down the road to python3/PyQt5.  I'm also moving to PiOS (bullseye) (PiOS starting 2021-10-30)

The upgrade path may be difficult for those on older versions of Raspbian/python2.   Not sure how to handle these issues yet, to make it easy to upgrade without starting over.

The python3/PyQt5 testing branch is Python3-SerBryden

/K

taftr1, 2022-02-18:
Using the latest version of Raspberry Pi OS (raspberrypi 5.10.92-v8+ #1514) I'm using the python3-SerBrynden Branch.  After working ot the changes between the install of the Master Branch and this one, I am getting the following when I attempt to do a First Start.   It is possible I missed something...   I did Blank out my Keys and location in this output

pi@raspberrypi:~/PiClock $ sh startup.sh -n -s
Disabling screen blanking....
Setting sound to max (assuming Monitor Tv controls volume)....
Checking for NeoPixels Ambilight...
1493
1496
Checking for GPIO Buttons...
Checking for Temperature Sensors...
Starting PiClock.... logging to screen.
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
map base url for radar1: http://maps.googleapis.com/maps/api/staticmap?key=KEYBLANKED&center=MYLAT,-MYLON&zoom=7&size=266x244&maptype=hybrid
map base url for radar2: http://maps.googleapis.com/maps/api/staticmap?key=KEYBLANKED&center=MYLAT,-MYLON&zoom=11&size=266x244&maptype=hybrid
map base url for radar3: http://maps.googleapis.com/maps/api/staticmap?key=KEYBLANKED&center=MYLAT,-MYLON&zoom=7&size=622x622&maptype=hybrid
map base url for radar4: http://maps.googleapis.com/maps/api/staticmap?key=KEYBLANKED&center=MYLAT,-MYLON&zoom=11&size=622x622&maptype=hybrid
libpng warning: iCCP: known incorrect sRGB profile
metar url: https://tgftp.nws.noaa.gov/data/observations/metar/stations/KFWN.TXT
getting current: Thu Feb 17 23:00:04 2022
https://api.climacell.co/v3/weather/realtime?apikey=KEYBLANKED&lat=MYLAT&lon=MYLON&unit_system=us&fields=temp,weather_code,feels_like,humidity,wind_speed,wind_direction,wind_gust,baro_pressure
getting hourly: Thu Feb 17 23:00:04 2022
https://api.climacell.co/v3/weather/forecast/hourly?apikey=KEYBLANKED&lat=MYLAT&lon=MYLON&unit_system=us&fields=temp,precipitation,precipitation_type,precipitation_probability,weather_code
getting daily: Thu Feb 17 23:00:04 2022
https://api.climacell.co/v3/weather/forecast/daily?apikey=KEYBLANKED&lat=MYLAT&lon=MYLON&unit_system=us&fields=temp,precipitation_accumulation,precipitation_probability,weather_code
wxstart for radar1
wxstart for radar2
Traceback (most recent call last):
  File "/home/pi/PiClock/Clock/PyQtPiClock.py", line 793, in wxfinished_cc
    wxdata = json.loads(wxstr)
  File "/usr/lib/python3.9/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.9/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.9/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
Aborted

SerBrynden, 2022-02-18:
Climacell.co changed to Tomorrow.io and their v3 API was replaced with the v4 API.  I don't think the v3 API works anymore and, last I checked, the PiClock code hasn't been updated to support the v4 API.  I use OpenWeatherMap.org.

taftr1, 2022-02-25:
So seeing 2 issues now, one is brand new.
The first is there seems to be some problem displaying the Radar Overlays.  The come up all garbled and mirrored is the best way I can describe it.  I am seeing "libpng warning: iCCP: known incorrect sRGB profile" running through the logs in sets of 3.

This morning a New error popped up.  This causes PiClock to Crash.  Its referencing "Compass" in the error.  Near as I can tell its something to do with decoding the METAR.  Originally I thought it had something to do with the METAR I was using not being available or missing.  However I was able to pull the file up and restarting PiClock caused the same issue.  I tried using a Different Airport code, utilizing the nearest international portrait, thinking its format would not have the same error.  However the same problem happens.  the code referenced is Line 1178 and is "wd = f.wind_dir.compass()"   Unfortunately my programing knowledge is limited here and cant seem to find where is defined.

libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
map base url for radar1: https://api.mapbox.com/styles/v1/bcurley/cj712peyz0bwr2sqfndbggupb/static/LAT,LON,6,0,0/266x244?access_token=TOKEN&attribution=false&logo=false
map overlay url for radar1: https://api.mapbox.com/styles/v1/bcurley/cj712r01c0bw62rm9isme3j8c/static/LAT,LON,6,0,0/266x244?access_token=TOKEN
map base url for radar2: https://api.mapbox.com/styles/v1/bcurley/cj712peyz0bwr2sqfndbggupb/static/LAT,LON,10,0,0/266x244?access_token=TOKEN&attribution=false&logo=false
map overlay url for radar2: https://api.mapbox.com/styles/v1/bcurley/cj712r01c0bw62rm9isme3j8c/static/LAT,LON,10,0,0/266x244?access_token=TOKEN
map base url for radar3: https://api.mapbox.com/styles/v1/bcurley/cj712peyz0bwr2sqfndbggupb/static/LAT,LON,6,0,0/622x622?access_token=TOKEN&attribution=false&logo=false
map overlay url for radar3: https://api.mapbox.com/styles/v1/bcurley/cj712r01c0bw62rm9isme3j8c/static/LAT,LON,6,0,0/622x622?access_token=TOKEN
map base url for radar4: https://api.mapbox.com/styles/v1/bcurley/cj712peyz0bwr2sqfndbggupb/static/LAT,LON,10,0,0/622x622?access_token=TOKEN&attribution=false&logo=false
map overlay url for radar4: https://api.mapbox.com/styles/v1/bcurley/cj712r01c0bw62rm9isme3j8c/static/LAT,LON,10,0,0/622x622?access_token=TOKEN
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
metar url: https://tgftp.nws.noaa.gov/data/observations/metar/stations/K12N.TXT
getting current and forecast: Fri Feb 25 10:04:33 2022
https://api.openweathermap.org/data/2.5/onecall?appid=KEY&lat=LAT&lon=LON&units=imperial&lang=en&r=0.4262510323217398
wxstart for radar1
wxstart for radar2
wxmetar: K12N 251454Z AUTO VRB05KT M01/M03 A2995 RMK AO1 SLP154 60017 T10111028 58037
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
Traceback (most recent call last):
  File "/home/pi/PiClock/Clock/PyQtPiClock.py", line 1178, in wxfinished_metar
    wd = f.wind_dir.compass()
AttributeError: 'NoneType' object has no attribute 'compass'
Aborted

taftr1, 2022-03-06:
Tried again today, and the Crashing and alert about "Compass" have stopped.  However the Image corruption continues

n0bel, 2022-03-06:
I don't see why it has that error.   K12N works fine for me in PiClock and PiClock3.   Perhaps an upgrade...

sudo pip3 install python-metar --upgrade

taftr1, 2022-03-08:
I ran the command you provided with the following results:

pi@raspberrypi:~ $ sudo pip3 install python-metar --upgrade
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: python-metar in /usr/local/lib/python3.9/dist-packages (1.4.0)
Collecting python-metar
  Using cached https://www.piwheels.org/simple/python-metar/python_metar-1.4.0-py3-none-any.whl (18 kB)

I restarted PiClock and it is working.  However this error seems to happen after its been running for a while.  I'll let you know should it fault again.   

I still see the problem with the radar overlays being garbled.  Its particularly visible on the second page where it looks like is random noise along with the radar overlays being mixed on both maps.  it seems to get worse over time with images from the previous update being mixed together with new update
[image]
s

taftr1, 2022-03-09:
Following up.
When the Metar changed to the following for Wind the crash stopped.

2022/03/09 19:54
K12N 091954Z AUTO **00000KT** 02/00 A2990 RMK AO1 SLP134 P0008 T00170000

taftr1, 2022-03-21:
So I opened a Issue with the Python Metar Github.  Seems that it is indeed an issue with that code.  They aren't quite sure yet how to fix it.  But there is a Work around...

If you Replace the 2 occurrences of:

wd = f.wind_dir.compass()

with:

if f.wind_dir is None:
   wd = " "
else:
   wd = f.wind_dir.compass()

That will set the wind direction to a blank in the cases that it isn't specified or VRB.  This will stop the crashing at least until a more elegant method is developed.

I am still having issues with the Radar overlay.  I'm not quite able to figure that out....

closed 2023-04-02 by n0bel