n0bel/PiClock #229: does not work in northern latitudes

xenon462, 2021-12-12:
ERROR: 
Traceback (most recent call last):
  File "PyQtPiClock.py", line 727, in wxfinished_ds
    wx.setText(f['summary'] + "\n" + s)
KeyError: 'summary'
Aborted

Error when using coordinates 68.96172, 33.10312
Now in the Northern latitudes is a polar night.
Sunrise will be in January 2022.
It used to be the same mistake.
Help me fix it.

https://github.com/n0bel/PiClock/issues/118

n0bel, 2021-12-12:
This is not the same is the error reported in #118.   In that issue the problem was retrieving 'sunRise' and 'sunSet' from the darksky api.   That problem is indeed fixed. 

The error you are reporting here is a different issue pulling the 'summary' data from an hourly or daily forecast.   For  some reason the api is not returning correct data.   There's not alot that can be done with missing data.  If you wanted to try adding...
```
if 'summary' in f:
```
ahead of the setText, it might work.  You'll still be missing summary forecast information, which is the root cause of the issue.

Since DarkSky is going away (was going to be by end 2021, now end 2022), I suggest you move to one of the other weather apis available in PiClock.

xenon462, 2021-12-13:
everything is clear. thanks for your help

closed 2022-03-06 by n0bel